ros2/launch #5: Add support for test processes

tfoote, 2015-08-03:
The launch library can be used to turn system tests into nosetests. However if the subcomponents of the system tests are either nosetests or gtests their xml output is not captured, only the return code is utilized.

rostest had a specific tag "test" which was aware that the subprocess was going to generate an xunit output passed appropriate flags to the subprocess and gathered up the results of the subprocess. We should support the same functionality to support system tests.

dirk-thomas, 2015-08-04:
We will wait with this until ros2/system_tests#14 can server as a use case.

wjwwood, 2019-03-22:
This is covered by the on going work in the new launch system, see: https://github.com/ros2/launch/issues/208

closed 2019-03-22 by wjwwood