ros2/launch #7: asyncio errors on Windows

wjwwood, 2015-08-31:
On my latest Windows job I'm seeing errors like this:

http://ci.ros2.org/job/ros2_batch_ci_windows/299/console
```
19:09:05 39: ======================================================================
19:09:05 39: ERROR: test_publisher_subscriber_cpp__nested__rmw_connext_dynamic_cpp__rmw_connext_cpp.test_publisher_subscriber
19:09:05 39: ----------------------------------------------------------------------
19:09:05 39: Traceback (most recent call last):
19:09:05 39:   File "C:\tools\python\lib\site-packages\nose\case.py", line 198, in runTest
19:09:05 39:     self.test(*self.arg)
19:09:05 39:   File "C:\Jenkins\workspace\ros2_batch_ci_windows\workspace\build\test_communication\test_publisher_subscriber_cpp__nested__rmw_connext_dynamic_cpp__rmw_connext_cpp.py", line 22, in test_publisher_subscriber
19:09:05 39:     rc = launcher.launch()
19:09:05 39:   File "C:\Jenkins\workspace\ros2_batch_ci_windows\workspace\install\Lib\site-packages\launch\launcher.py", line 57, in launch
19:09:05 39:     asyncio.get_event_loop_policy().set_child_watcher(None)
19:09:05 39:   File "C:\tools\python\lib\asyncio\events.py", line 486, in set_child_watcher
19:09:05 39:     raise NotImplementedError
19:09:05 39: nose.proxy.NotImplementedError: 
19:09:05 39: -------------------- >> begin captured stdout << ---------------------
```

I think it might be related to this change?: https://github.com/ros2/launch/commit/368422353e281844c0441b0167e586a660ea42d9#diff-64876ece93fadad6549af3a8bcb112c7R57

I'm not sure though.

dirk-thomas, 2015-08-31:
Can you please try it again with #8.

tfoote, 2015-08-31:
The ci finished with a lot of test failures. (83)

dirk-thomas, 2015-08-31:
The problem is that the not succeeding tests are already there before: e.g. in http://ci.ros2.org/job/ros2_batch_ci_windows/299/

But since the nosetests are marked as "errors" and not "failures" the Jenkins xUnit plugin (configured for GoogleTests) does not pick them up as failing tests.

Therefore I would propose to merge #8 since it clearly fixes to invoke a not existing method on Windows. And after that the tests still need to be fixed.

dirk-thomas, 2015-08-31:
Since #8 only fixed the reporting (by not having tests with "errors" but "failures") I will reopen this.

dirk-thomas, 2015-08-31:
The current state is getting build in: http://ci.ros2.org/job/ros2_batch_ci_windows/309/

dirk-thomas, 2015-09-01:
Closed by #9.

commit f8c6a9d9 (2015-08-31): only unset watcher when not on Windows (fix #7)
commit f7f88727 (2015-08-31): Merge pull request #8 from ros2/intra_process_demo_fixups

only unset watcher when not on Windows (fix #7)
closed 2015-09-01 by dirk-thomas