ros2/launch #26: launch_testing always ignores license output when checking regexes

dhood, 2016-04-19:
`launch_testing` has support for filtering the RTI license output when checking that the received output matches the expected output. If the license output appears and is not being filtered, then the check fails. I think that this is the desired behaviour.

However, this concept seems to be implemented differently for the case when `regex_match` is `True`. The license output is always ignored, whether or not it was supposed to be. (And any other output that appears that doesn't match the regex)

These non-matching, un-filtered lines are written [here](https://github.com/ros2/launch/blob/master/launch_testing/launch_testing/__init__.py#L57) and read [here](https://github.com/ros2/launch/blob/master/launch_testing/launch_testing/__init__.py#L80), but since `self.matched` is `True`, the assertion does not fail.

Am I right in interpreting this as a bug that I should fix?

dhood, 2016-04-19:
this could also be addressed by adding `^` to the start of regexes (https://github.com/ros2/demos/blob/master/intra_process_demo/test/two_node_pipeline.regex for example), but that leads to test timeouts instead of failures

dirk-thomas, 2016-04-19:
The matrix of what to test for and how to test for it (https://github.com/ros2/launch/pull/13#discussion_r44998991) was implemented incompletely in that PR. I agree with your conclusion that this should be fixed by making the different tests do the same thing.

dhood, 2016-04-19:
thanks for that link! OK good, not a bug, just not yet implemented. 
I'll work on getting it to fail if lines which aren't covered by the regex are detected. (**B1**, I believe)

dhood, 2016-04-21:
I've started implementing this. At the moment, any lines that are not filtered and are not matched by the regex cause the test to fail.

This affects `ros2/demos` (pinging @jacquelinekay) because there are currently some test regexes which do not include all output lines, e.g.:
http://ci.ros2.org/job/ci_linux/1216/testReport/%28root%29/test_pendulum__rmw_connext_cpp/test_executable/

In this example case, do you think it's best to:
1. change the test regexes to expect the `'Logger node initialized.'` line
2. leave the regex as it is, but change the executable output to be something like `[debug] 'Logger node initialized.'` and then add an `[debug]` prefix filter to the test (or somehow use a different logging level)
3. add an option in `launch_testing` to ignore any lines that don't match the regex, as long as at some point the regex is matched

thanks

sloretz, 2018-03-01:
> At the moment, any lines that are not filtered and are not matched by the regex cause the test to fail.

@dhood Did your questions get resolved? Is this something that should be solved before bouncy? I see the test linked in your post `test_pendulum__rwm_connext_cpp` is being [skipped on the latest nightly](https://ci.ros2.org/view/nightly/job/nightly_linux_release/735/testReport/(root)/pendulum_control/test_pendulum__rmw_fastrtps_cpp_xunit_missing_result/). Is it skipped because of this issue?

dhood, 2018-03-01:
the pendulum_control package only builds with connext, it's skipped for fastrtps (for reasons not related to output matching). I have https://github.com/ros2/launch/pull/34/files that doesn't let tests pass if unexpected output occurs even with regex matching, but in the end it would have caused more tests to be flaky because we are relying on loose output matching in some places (e.g. composition tests). I remember an opt-out option which if applied to those situations might make that PR reasonable again.. I'll have to revisit it and decide if it's worth reviving or dropping.

wjwwood, 2019-03-22:
Closing since the legacy launch testing is no longer in use.

closed 2019-03-22 by wjwwood