ros2/launch #31: integrated and automated launch on OS boot

ahundt, 2016-05-09:
I'm proposing/requesting a way for a launch file or launch command to automatically be run when the computer boots, which is fully integrated and automated via ros2/launch.

Based on conversation at:
https://groups.google.com/forum/#!topic/ros-sig-ng-ros/0HzF5J6hu4k

Here is one proposal for how this could work, although alternative designs are worth considering:

#### `roslaunch <my_params> --add-at-startup`

expected behavior:

1. launching may or may not occur right away (tbd)
2. shut down machine
3. turn on machine
4. launch command runs

#### `roslaunch --list-at-startup`

expected behavior:

prints a list of commands that run every time the machine starts (perhaps with numeric ids?)

#### `roslaunch --remove-at-startup <id>`

expected behavior:

remove the item with id <id> from the startup list.

#### <lauch_at_startup>

a tag that goes in .launch files, whenever the launch is run this automatically adds it to the startup list

tfoote, 2016-05-09:
Can you explain why you want this to be built into the launch functionality instead of using the built in startup solutions such as systemd, upstart, and launchd? There are over a dozen existing startup tools: https://en.wikipedia.org/wiki/Init#Replacements_for_init also known as System Service Management tools: https://en.wikipedia.org/wiki/Operating_system_service_management

In general we try to design things to be small and self contained, and to leverage existing tools which already do the job. A tool for supporting launching on boot could be very useful, but I would suggest that it should be separate from the launching capabilities, and it should leverage the launching tool to make actually launch the processes. 

The process for registering commands to be run at startup is very platform and distro specific. And sometimes varies between versions. I think that an implementation of integrating ros nodes at startup would be better if it was tailored to specific service management implementation. For example there's already a package [robot_upstart](http://wiki.ros.org/robot_upstart) which supports integration with upstart. 

This clearly separates the roles of launching (process management) and configuring startup scripts(setting, modifying, and reading startup configuration). The launching can be used across all platforms while the startup configuration tools may be platform or init tool specific.

ahundt, 2016-05-10:
> Can you explain why you want this to be built into the launch functionality instead of using the built in startup solutions such as systemd, upstart, and launchd? There are over a dozen existing [...]

> In general we try to design things to be small and self contained, and to leverage existing tools which already do the job. A tool for supporting launching on boot could be very useful, but I would [...]

Perhaps a rescoping of roslaunch for ROS 2 would be sensible, because if you're deploying a real system roslaunch is necessary but simply isn't sufficient. Nodes can crash or stop communicating and need to recover from failure somehow. Therefore, this issue combined with #32 and #33 suggests making the ROS 2 version of roslaunch be a small self contained node lifetime management tool. Of course, one necessary step for a lifetime management tool is to start it.

I think fundamentally the idea is that all robots have to start, and ideally they should be able to do so on their own. Therefore I'm proposing such functionality as a reasonable thing for ros/ros2 to support and make super easy to use in the default installation. It could be somewhere else, I simply posted here based on a suggestion from the list linked in my first post.

I'm definitely wide open when it comes to how this is implemented. Would make the most sense for whatever provides this functionality to use these the tools you mention under the hood?

Perhaps alternately it would be best to just have setup scripts that work on a few supported platforms that initialize this utility at startup?

> The process for registering commands to be run at startup is very platform and distro specific. And sometimes varies between versions. I think that an implementation of integrating ros nodes at startup would be better if it was tailored to specific service management implementation. 

Could such a thing respond to messages telling it to shut down?

> For example there's already a package robot_upstart which supports integration with upstart.

Cool, looks like a decent tool but perhaps ubuntu only?

It looks like qt has [qtservice](https://github.com/qtproject/qt-solutions/tree/master/qtservice), a slightly stale tool of a similar nature, though if it depends on other parts of qt that's quite a large dependency.

The discussion was also about launching being done via python, so perhaps tools/classes exist that will work. [python-daemon]( https://pypi.python.org/pypi/python-daemon/) looks like a Unix only example

asmodehn, 2016-05-10:
Just my two cents, coming from the server world, where "always running" services in small containers are a necessity.

There are already quite a few cross platform process managers out there, but I haven't seen them mentioned in the roslaunch discussion yet.
One of the most popular python tool currently might be http://supervisord.org/ but each has is own specific features (like https://circus.readthedocs.io/en/latest/ that also manages sockets for example).

I'm not sure how these compare with roslaunch (ROS1 or ROS2) feature wise, but I really wonder what would a "ROS-specific" process manager implement that is not already (and cannot be) done by a generic process manager...

After all, it seems to me that this problem is a generic software problem, and not specific to ROS.

dirk-thomas, 2016-05-10:
@ahundt I agree with @tfoote's statement above that `roslaunch` should focus on one task (starting, restarting, checking processes). A separate tool can be responsible for registering something for startup. Obviously that tool can call / reuse roslaunch to delegate most of the work.

@asmodehn The launching capability in ROS 2 should also include verifying that set of processes has started correctly, the communication graph between them is as expected and handle different error cases (process dies, connection lost, etc.). Starting the processes is only a very tiny part of the implementation. The majority will be designing an API to make launch files (written in Python) still introspectable, performing the life cycle management of the nodes and other ROS specific functionality. We will take a closer look to existing tools but I assume that they won't help us in developing all this ROS specific functionality in `launch `

ahundt, 2016-05-10:
> @ahundt I agree with @tfoote's statement above that roslaunch should focus on one task (starting, restarting, checking processes). A separate tool can be responsible for registering something for startup (DOTADIW). Obviously that tool can call / reuse roslaunch to delegate most of the work.

That sounds like a good design, particularly if it is easy to start roslaunch, or if roslaunch has a config file or some small necessary part that may be necessary for the other tool to work with it properly.

Ultimately, I think what really matters is to ensure people can make their robot boot automatically by following a simple how-to for various OSes in a single simple document that comes with the ROS2 software, not necessarily my first specific suggestion at the top.

sloretz, 2018-03-01:
Hi @ahundt, It sounds like you may be interested in the design doc for ROS 2's launch system being worked on at ros2/design#163. It is very early and very incomplete, but it would be a good place to watch add comments as it develops.

I'll close this issue because I don't expect it to get work on in this repository; however it can be reopened later if that turns out not to be the case.

closed 2018-03-01 by sloretz

ahundt, 2018-03-11:
sounds good, thanks!