ros2/launch #32: integrated, automated, and programmable monitoring + relaunch

ahundt, 2016-05-09:
I'm proposing/requesting a way for roslaunch to detect failures and restart components using reasonable defaults and programmable parameters. 

This idea probably needs some refinement but here is my first pass based on the conversation at:
https://groups.google.com/forum/#!topic/ros-sig-ng-ros/0HzF5J6hu4k

Proposed functionality:

#### roslaunch itself runs as a daemon that is perpetually up once initially launched

This bears some similarities for roscore in ROS 1, but it is for a different purpose.

Launch files specify expected behavior of the nodes, and roslaunch verifies that those conditions are met. If the conditions are violated the node is automatically killed and relaunched.

For example, a minimum and maximum frequency of messages published on specified topics, and perhaps a heartbeat message that exists and is enabled by default. If the heartbeat fails or publishing rates exceed the max then the node is killed and relaunched.

All nodes would have this behavior integrated and enabled by default with reasonable default parameters that work for 90% of users. Assuming launch files are python (I think this was discussed in the mailing list) when other requirements exist this functionality can be disabled or modified with explicit changes to the launch file code.

Note that the roslaunch command can still be used exactly as it is now by users. The only difference is that when run, a new roslaunch process would look for an existing running daemon and if one exists, it would pass its parameters to the perpetual daemon to perform the actual launching.

##### sub-proposal 1: roslaunch daemon automatic startup at OS boot

The roslaunch daemon is supplied as one of the items to launch at OS boot by default https://github.com/ros2/launch/issues/31

If not by default, I suggest a simple one line command which enables booting at OS launch. This should hopefully be cross platform, i.e. work for ubuntu, linux, mac, and perhaps other OSes.

##### sub-proposal 2: roslaunch daemon lifetime

The roslaunch daemon should detect its own crashes and/or death and automatically restart itself, except when a user runs something such as `roslaunch shutdown` which tells it to exit permanently.

This could be done by an extremely simple micro-daemon or script which only manages roslaunch.

dirk-thomas, 2016-05-10:
The features you describe are all very important and we plan to implement most of them.

I am not sure that launch needs to be realized as a daemon in all those case. But when a launch file is already "running" it should be possible to add service interface to it in order to introspect the state as well as manage its processes. I would think it will become a hybrid of the functionality available in roslaunch and nodelet managers in ROS 1.

As mentioned in the other ticket: the aspect of starting something like launch during startup should be handled separately ([DOTADIW](https://en.wikipedia.org/wiki/Unix_philosophy#Do_One_Thing_and_Do_It_Well)).

ahundt, 2016-05-10:
Sounds awesome! I also read the [ROS 2 document about node lifecycles](http://design.ros2.org/articles/node_lifecycle.html
) which is worth mentioning here.

Perhaps will an equivalent proposal be needed for the tool which manages the lifecycles, which may be be what `ros2/launch` ends up doing?

sloretz, 2018-03-01:
Hi @ahundt, It sounds like you may be interested in the design doc for ROS 2's launch system being worked on at ros2/design#163. It is very early and very incomplete, but it would be a good place to watch add comments as it develops.

I'll close this issue because I don't expect it to get work on in this repository; however it can be reopened later if that turns out not to be the case.

closed 2018-03-01 by sloretz

ahundt, 2018-03-11:
sounds good, thanks!