ros2/launch #33: roslaunch daemon as an update tool

ahundt, 2016-05-09:
I'm proposing/requesting a way for a roslaunch daemon to automate software updates

This is dependent on #32, and related to #31.

This idea probably needs some refinement but here is my first pass based on the conversation at:
https://groups.google.com/forum/#!topic/ros-sig-ng-ros/0HzF5J6hu4k

The roslaunch daemon would provide a facility where it receives an update message for a specific component or set of components. The daemon would take down all relevant nodes (or potentially all running nodes), then run a command which downloads/updates packages. (perhaps rosdep update?)

It should be possible to support updates from forks of ros, and provide packages on a local network and/or on a USB stick. 

If possible, I would suggest these updates be verified with some sort of signature from the start.

Once the update is completed, which may include updates to the list of roslaunch commands proposed in #31, all nodes which are down should be automatically relaunched.

dirk-thomas, 2016-05-10:
While the feature is certainly useful I would use the same arguments as in other tickets. `launch` should focus on one task (starting, restarting, checking processes) ([DOTADIW](https://en.wikipedia.org/wiki/Unix_philosophy#Do_One_Thing_and_Do_It_Well)). A separate tool can be responsible to for the updating part and the orchestration. But `launch` should be designed in such a way that it can provide the necessary interfaces to support this use case:

* tell launch to stop a set of processes
* perform an update of the packages (which will be very platform specific)
* tell launch to restart a set of processes

ahundt, 2016-05-10:
Your suggestion makes perfect sense to me that sounds like a great solution

sloretz, 2018-03-01:
Hi @ahundt, It sounds like you may be interested in the design doc for ROS 2's launch system being worked on at ros2/design#163. It is very early and very incomplete, but it would be a good place to watch add comments as it develops.

I'll close this issue because I don't expect it to get work on in this repository; however it can be reopened later if that turns out not to be the case.

closed 2018-03-01 by sloretz

ahundt, 2018-03-11:
sounds good, thanks!