ros2/launch #37: Environment passing to launch has no effect

mikaelarguedas, 2016-07-05:
Spawned processes always get the parent's environment even though a specific env parameter is given.

Reproducible example [here](https://gist.github.com/mikaelarguedas/85a7eebb1dafe1a73c3b26d95ca075e3)

dirk-thomas, 2016-07-05:
The `env` parameter of `add_process` is not being used anywhere at the moment. If you need to pass custom environments you need to make it pass this along to the subprocess.

mikaelarguedas, 2016-07-05:
I see. I discovered this while testing https://github.com/ros2/system_tests/blob/master/test_communication/test/test_publisher_subscriber.py.in which I think is the only place that uses it for now.
Just to confirm my understanding, the purpose of the `env` parameter is indeed to pass custom environment but doesn't pass them to the spawned subprocesses yet, is that what you mean ?

dirk-thomas, 2016-07-05:
Yes, the function has such a parameter but it is not applied anywhere.

closed 2016-07-12 by mikaelarguedas