ros2/launch #39: Using types for process vs co-routine checking

rohbotics, 2016-07-27:
Right now there is a lot of this in the launcher code:
```
if 'protocol' in dir(p):
    do stuff for process
if 'coroutine' in dir(p):
    do stuff for coroutine 
```

Does it make sense to do something like this instead:
```
if isinstance(p, ProcessDescriptor):
    do stuff for process
if isinstance(p, CoroutineDescriptor):
    do stuff for coroutine 
```

Or does that not canonical python?

dirk-thomas, 2016-07-27:
The first block is imo the more common approach in Python. Often a function / attribute is just being used and wrapped in a try / except block. See https://en.wikipedia.org/wiki/Duck_typing#In_Python

wjwwood, 2016-07-27:
The first is slight more flexible, and so may be preferable. However, I think [`hasattr`](https://docs.python.org/3/library/functions.html#hasattr) is probably a more readable way to check if the given object has a particular function or member and therefore meets the required interface.

dirk-thomas, 2016-10-03:
Since this ticket was more a questions which has been answered I will close this. If anyone would like to change the condition please create a PR for it.

closed 2016-10-03 by dirk-thomas