ros2/launch #41: No exit_handler to check the error code of a program that closes

geoffviola, 2016-09-26:
Inside the https://github.com/ros2/launch/blob/master/launch/launch/exit_handler.py file, all the exit handlers ignore output, restart the program, or run the default_exit_handler, which shuts all the programs down. I want to launch a program and check its return code for failure. If the error code is non-zero, the launch file should tear itself down. Is that in line with the current design?

dirk-thomas, 2016-09-26:
I am not sure I understand your exact use case and how it diverges from what is possible with the existing  exit handlers but the design aims to enable implementing any logic using custom handlers.

geoffviola, 2016-09-26:
Here's the scenario. I have a launch file that starts a node and a custom parameter client. The parameter client will set the parameters and return with a zero exit code, if it ran properly. I do want the launch file to tear down if the error code is non-zero, because I don't want to run the node without the correct parameters. Currently, no exit_handler supports this scenario.

dirk-thomas, 2016-09-26:
Likely we haven't had this use case in any of the exiting demos and examples yet. A custom written exit handler should be able to fulfill this use case. You can either keep it in your package or create a pull request to contribute it to the `launch` package.

clalancette, 2017-08-04:
I'm going to close this out since I closed out the PR that implemented this.  Feel free to re-open if we want to keep this feature request.

closed 2017-08-04 by clalancette