ros2/launch #56: Launch doesn't always bring down all nodes on keyboard interrupt

dhood, 2017-06-22:
I have a launch script that starts various ROS 2 nodes. Most times if I kill it with ctrl-c it dies gracefully with no issues. Occassionally (I haven't figured out under what circumstances yet) it gives:

```
^CTraceback (most recent call last):
  File "/home/ubuntu/tb2_demo_ws/build_isolated/launch/launch/launcher.py", line 91, in launch
    returncode = loop.run_until_complete(generator)
  File "/usr/lib/python3.5/asyncio/base_events.py", line 375, in run_until_complete
    self.run_forever()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 345, in run_forever
    self._run_once()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 1276, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.5/selectors.py", line 441, in select
    fd_event_list = self._epoll.poll(timeout, max_ev)
KeyboardInterrupt

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/tb2_demo_ws/install_isolated/launch/bin/launch", line 9, in <module>
    load_entry_point('launch', 'console_scripts', 'launch')()
  File "/home/ubuntu/tb2_demo_ws/build_isolated/launch/launch/main.py", line 47, in main
    rc = launcher.launch()
  File "/home/ubuntu/tb2_demo_ws/build_isolated/launch/launch/launcher.py", line 99, in launch
    loop.run_forever()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 345, in run_forever
    self._run_once()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 1276, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.5/selectors.py", line 441, in select
    fd_event_list = self._epoll.poll(timeout, max_ev)
KeyboardInterrupt
Exception ignored in: <bound method BaseEventLoop.__del__ of <_UnixSelectorEventLoop running=False closed=True debug=False>>
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/base_events.py", line 431, in __del__
  File "/usr/lib/python3.5/asyncio/unix_events.py", line 58, in close
  File "/usr/lib/python3.5/asyncio/unix_events.py", line 139, in remove_signal_handler
  File "/usr/lib/python3.5/signal.py", line 47, in signal
TypeError: signal handler must be signal.SIG_IGN, signal.SIG_DFL, or a callable object
Exception ignored in: <bound method BaseSubprocessTransport.__del__ of <_UnixSubprocessTransport closed pid=1247 running stdin=<_UnixWritePipeTransport closing fd=23 open> stdout=<_UnixReadPipeTransport fd=24 open>>>
Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/base_subprocess.py", line 126, in __del__
  File "/usr/lib/python3.5/asyncio/base_subprocess.py", line 101, in close
  File "/usr/lib/python3.5/asyncio/unix_events.py", line 568, in close
  File "/usr/lib/python3.5/asyncio/unix_events.py", line 560, in write_eof
  File "/usr/lib/python3.5/asyncio/base_events.py", line 497, in call_soon
  File "/usr/lib/python3.5/asyncio/base_events.py", line 506, in _call_soon
  File "/usr/lib/python3.5/asyncio/base_events.py", line 334, in _check_closed
RuntimeError: Event loop is closed
```

On at least one occasion this has left one of the nodes running (as was happening in https://github.com/ros2/launch/issues/54 ).

I've seen this issue https://mail.python.org/pipermail/python-bugs-list/2016-January/290450.html that suggests that the output isn't necessarily something to be concerned about, but perhaps in our use case it prevents the remaining nodes from being killed.

I know that this is vague: I will try to add more information about how to reproduce this when I find out.

dhood, 2017-06-29:
I think that the issue was occurring because by habit I press ctrl-c twice. So, I was interrupting the teardown. I'll close this since it seems like user error on my part.

wjwwood, 2017-07-17:
> I think that the issue was occurring because by habit I press ctrl-c twice. So, I was interrupting the teardown. I'll close this since it seems like user error on my part.

It might be user error, but it's still something I imagine people will do. Ideally it would not leak processes no matter how many times you press ctrl-c. `roslaunch` from ROS 1 handles this correctly I believe. You can only get leaked processes if you SIGKILL it with ctrl-\.

dhood, 2017-07-19:
ok, let's reopen this then if this should be made user-proof

PR #58 (2017-08-12): suppress repeated SIGINTs in order to cleanly shutdown
Fixes #56.

After the first `KeyboardInterrupt` has been catched suppress any further SIGINTs. This will ensure that all processes are cleaned up correctly. The user can still choose to `Ctrl+\` (rather than `Ctrl+C`).
closed 2017-08-12 by dirk-thomas