ros2/launch #62: Launch files broken since #57 has been merged

mikaelarguedas, 2017-09-11:
Opening this for visibility
I tried the lifecycle launchfile and the dummy_robot_bringup one and they both crash. If I revert #57 they work.

dirk-thomas, 2017-09-11:
I can confirm that the current state is broken. Are you going to work on a fix or should I look into it?

mikaelarguedas, 2017-09-11:
I'll look into it

wjwwood, 2017-09-11:
@mikaelarguedas can you please post the error you get in the crash, so others can confirm it's the same issue. I get:

```
% launch share/dummy_robot_bringup/launch/dummy_robot_bringup.py
Traceback (most recent call last):
  File "/tmp/testing_beta3/bin/ros2-osx/bin/launch", line 11, in <module>
    load_entry_point('launch==0.0.0', 'console_scripts', 'launch')()
  File "/tmp/testing_beta3/bin/ros2-osx/lib/python3.6/site-packages/launch/main.py", line 50, in main
    load_launch_file(args.launch_file, launch_descriptor, args.args)
  File "/tmp/testing_beta3/bin/ros2-osx/lib/python3.6/site-packages/launch/loader.py", line 20, in load_launch_file
    launch_module = loader.load_module()
  File "<frozen importlib._bootstrap_external>", line 399, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 823, in load_module
  File "<frozen importlib._bootstrap_external>", line 682, in load_module
  File "<frozen importlib._bootstrap>", line 245, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 427, in spec_from_loader
  File "<frozen importlib._bootstrap_external>", line 574, in spec_from_file_location
  File "<frozen importlib._bootstrap_external>", line 664, in is_package
  File "<frozen importlib._bootstrap_external>", line 66, in _path_split
AttributeError: 'list' object has no attribute 'rpartition'
```

dirk-thomas, 2017-09-11:
Same for me.

mikaelarguedas, 2017-09-11:
Same for me

mikaelarguedas, 2017-09-11:
fixed in #63

closed 2017-09-11 by mikaelarguedas