ros2/launch #64: Launcher hangs with keyboard interrupt on Windows

dhood, 2017-09-13:
Most of our usage of launch files is in tests, where launch_testing causes signals to be sent. But if you run them manually and ctrl-c them, they hang (for example `demo_nodes_cpp`'s `test_set_and_get_parameters__rmw_fastrtps_cpp_Release.py`). I'm yet to find a launch file that doesn't cause this to happen. 

In https://github.com/ros2/demos/issues/175#issuecomment-329030173 @dirk-thomas noted that 
> The process hangs for me in `launcher.py` on the `loop.close()` call. Both subprocesses have been `SIGTERM`ed and the loop is not running anymore. But the `close()` call still hangs forever.

Simple ROS2-agnostic example:

```
$ cat launch_wait.py
import os

from launch import LaunchDescriptor
from launch.launcher import DefaultLauncher

def main():
    launcher = DefaultLauncher()
    launch_descriptor = LaunchDescriptor()

    if os.name == 'nt':
        cmd=['ping', '127.0.0.1', '-n', '60']
    else:
        cmd=['sleep', '60']
    launch_descriptor.add_process(
        cmd=cmd,
        name='dummy_process',
    )
    launcher.add_launch_descriptor(launch_descriptor)

    rc = launcher.launch()
    print('Return code: ' + str(rc))
    exit(rc)

if __name__ == '__main__':
    main()
```
Behaviour on linux:
```
$ python3 launch_wait.py
(dummy_process) pid 13687: ['sleep', '60'] (stderr > stdout, all > console)
^C(dummy_process) signal SIGINT
(dummy_process) rc -2
Return code: 1
```

Behaviour on windows:
```
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC>python C:\Users\osrf\Downloads\r2b3-testing\overlay\src\launch_nothing.py
(dummy_process) pid 4416: ['ping', '127.0.0.1', '-n', '60'] (stderr > stdout, all > console)
[dummy_process]
[dummy_process] Pinging 127.0.0.1 with 32 bytes of data:
[dummy_process] Reply from 127.0.0.1: bytes=32 time<1ms TTL=128
[dummy_process] Reply from 127.0.0.1: bytes=32 time<1ms TTL=128
[dummy_process] Reply from 127.0.0.1: bytes=32 time<1ms TTL=128
<manually ctrl c>
(dummy_process) signal SIGTERM
(dummy_process) rc 3221225786
<hangs>
```

dhood, 2017-09-13:
Clarification: this is referring to python scripts that create their own launcher manually and are invoked with `python` directly.

As @wjwwood noted in https://github.com/ros2/demos/issues/175#issuecomment-329030173 this does NOT happen for "launch files" that are invoked with `launch`, e.g. https://github.com/ros2/demos/blob/a4d48aa57ba5752122c23bf79e0959052b96337e/lifecycle/launch/lifecycle_demo_launch.py

dhood, 2017-09-13:
I'm at a loss as to why converting the example into the same format as https://github.com/ros2/demos/blob/a4d48aa57ba5752122c23bf79e0959052b96337e/lifecycle/launch/lifecycle_demo_launch.py doesn't fix it.

This still hangs (executed with `launch` rather than `python`) on Windows:
```
import os

from launch.exit_handler import primary_exit_handler

def launch(launch_descriptor, argv):
    
    if os.name == 'nt':
        cmd=['ping', '127.0.0.1', '-n', '60']
    else:
        cmd=['sleep', '60']
    launch_descriptor.add_process(
        cmd=cmd,
    )
    launch_descriptor.add_process(
        cmd=cmd,
        exit_handler=primary_exit_handler,
    )
```

I have on _one_ occasion seen it exit cleanly so there might be a race condition involved.

rhololkeolke, 2018-01-21:
I have this same issue on Linux. It seems that `loop.run_forever` is called, but `loop.stop` is never called.

If you add a `loop.stop` to the end of `DefaultLaunch._run`  it fixes the issue. However, I'm not sure if it causes any other issues. I'm not super familiar with the asyncio library.

sloretz, 2018-03-01:
FYI for future readers, #69 fixed on OSX, but issue still exists on windows.

wjwwood, 2019-03-22:
Closing since the legacy launch testing is no longer in use.

closed 2019-03-22 by wjwwood