ros2/launch #65: API and doc

oroulet, 2017-10-17:
Hi,
I was trying to find out how to use this launch script. After some googling I found this:  https://github.com/ros2/turtlebot2_demo/blob/master/turtlebot2_teleop/launch/turtlebot_joy.py
With the following example:

```
def launch(launch_descriptor, argv):
    ld = launch_descriptor
    package = 'turtlebot2_drivers'
    ld.add_process(
        cmd=[get_executable_path(package_name=package, executable_name='kobuki_node')],
        name='kobuki_node',
        exit_handler=restart_exit_handler,
    )
package = 'teleop_twist_joy'

```

Is this the final API? It looks quite heavy. The first thing I did was to make my own function:

```
def add(ld, pkg, name, exit_on_die=True):
    if exit_on_die:
        handler = restart_exit_handler,
    else:
        handler = default_exit_handler
    ld.add_process(
        cmd=[get_executable_path(package_name=pkg, executable_name=name)],
        name=name,
        exit_handler=handler
    )

```

So this bug request is about two things
1- If this is the expected API/syntax in launch file. Can I update README/Wiki to document it?
2- If this is not the final API, what will it be? What is missing?

Thanks

dirk-thomas, 2017-10-17:
> Is this the final API?

No, it only represents the current development state.

> If this is not the final API, what will it be? What is missing?

Cleaner API, introspective capabilities, potentially a XML frontend for the "simple" use cases...

oroulet, 2017-10-17:
Where is the discussions on design like this happening? Discourse? I am trying to evaluate the use of ros2 for a robot going out in a few years, it is quite hard to find out the state of the core modules (we do not need other nodes yet). Not sure where to ask.

dirk-thomas, 2017-10-17:
Currently nobody is actively working on `launch` (see [roadmap](https://github.com/ros2/ros2/wiki/Roadmap)). Usually the discussion of the concepts is being done in an issue or PR in the [design](https://github.com/ros2/design/) repo.

dirk-thomas, 2017-10-17:
> What about a run method defined in launcher.py that matches the ros2 command?

That seems to only expose a part of the capabilities. I don't know how useful that would be. Please feel free to create a PR for it if it would support your use case.

oroulet, 2017-10-20:
Looking at code and use cases

closed 2017-10-20 by oroulet