ros2/launch #66: Default exit handler always has return code of 0 on windows

dhood, 2017-12-06:
In https://github.com/ros2/rcl/pull/198 I tried to add a test that checks for non-zero return codes for a particular process. On windows, the process would have a non-zero return code, but the launch would always have a return code of 0.

I think, using the default exit handler, it's not possible for the return code to ever be set on windows, because this line:
https://github.com/ros2/launch/blob/a1057c703a36a43963e1971bcb1b36ccff372d85/launch/launch/exit_handler.py#L43

combined with the line above that always ensures that `context.launch_state.teardown` is `True`:
https://github.com/ros2/launch/blob/a1057c703a36a43963e1971bcb1b36ccff372d85/launch/launch/exit_handler.py#L36

means that the branch where the return code of the launch is set can never be entered.

mikaelarguedas, 2017-12-06:
Based on the comment
https://github.com/ros2/launch/blob/a1057c703a36a43963e1971bcb1b36ccff372d85/launch/launch/exit_handler.py#L41-L42
It may be worth looking at the behavior with python 3.5+ (we're currently using 3.6.3 on the windows machines and require 3.5 for all our python code)

dhood, 2018-05-04:
@sloretz I just stumbled upon this issue which doesn't have the bouncy milestone anymore but it doesn't seem like the milestone was ever intentionally removed (maybe a ui bug?). shall we put the bouncy milestone back manually, or did you actually remove it?

sloretz, 2018-05-04:
@dhood I don't remember. It doesn't look like this issue was ever intentionally put into the bouncy milestone.

I just added and removed the milestone to see what the github UI says. It looks like the github ui says "added" when a milestone is added, and "cleared" when it's removed. However if those happen back-to-back it collapses them both into "modified". I think on Mar 1 I must have accidentally added this to the bouncy milestone, then cleared it and set labels instead.

wjwwood, 2019-03-22:
Closing since the legacy launch testing is no longer in use.

closed 2019-03-22 by wjwwood