ros2/launch #79: Multi-machine Launching

matt-attack, 2018-06-15:
## Feature request

#### Feature description
Add a way to launch processes on multiple machines using ssh in a fashion similar to the old launch system. Would allow for the launching of more complex systems.

wjwwood, 2018-06-20:
Thanks for the feature request, it's on our todo list, but not something we'll probably do in the near future on our own.

m-morelli, 2021-03-24:
Hello everyone, which is the status of this feature (also mentioned [here](https://github.com/ros2/ros2/issues/734#issuecomment-524114915))? Is this already supported? I couldn't find any doc about it...
Thanks!

hidmic, 2021-03-24:
@m-morelli to the best of my knowledge, execution across multiple machines is still a work in progress. See https://github.com/ros2/design/pull/297 and https://github.com/ros2/design/pull/272.

DanyResasco, 2022-05-23:
hello, what's the status of this feature? would be very nice to have it.

elandini84, 2023-05-02:
Does anyone know anything about the status of the feature?

vknisley-swri, 2023-12-19:
I'm looking into implementing this. I've discussed this with / am in communication with those who worked on the designs linked above (@mlanting and @roger-strain).

vknisley-swri, 2024-07-26:
Apologies, as it turns out, I will not be able to continue work on adding this feature in the near future. Hopefully someone else can take it on.