ros2/launch #89: Launching on arm64 with Fast-RTPS with fat archive from 2018-06-21 never quits

clalancette, 2018-06-21:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Bionic, arm64
- Installation type:
  - Fat archive from 2018-06-21, downloaded from here: https://ci.ros2.org/view/packaging/job/packaging_linux-aarch64/475/artifact/ws/ros2-package-linux-aarch64.tar.bz2
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
$ ros2 launch share/demo_nodes_cpp/launch/topics/talker_listener.launch.py
```

Let it run for a few seconds, then Ctrl-C:

```
root@changeling:~/ros2-linux# ros2 launch share/demo_nodes_cpp/launch/topics/talker_listener.launch.py
[INFO] [launch]: process[talker-1]: started with pid [21936]
[INFO] [launch]: process[listener-2]: started with pid [21937]
[INFO] [talker]: Publishing: 'Hello World: 1'
[INFO] [listener]: I heard: [Hello World: 1]
[INFO] [talker]: Publishing: 'Hello World: 2'
[INFO] [listener]: I heard: [Hello World: 2]
[INFO] [talker]: Publishing: 'Hello World: 3'
[INFO] [listener]: I heard: [Hello World: 3]
^C[WARNING] [launch.LaunchService]: user interrupted with ctrl-c (SIGINT)
signal_handler(2)
signal_handler(2)
[INFO] [launch]: process[talker-1]: process has finished cleanly
[INFO] [launch]: process[listener-2]: process has finished cleanly
```

It hangs there forever, until the user hits Ctrl-C.

#### Expected behavior
Launch file terminates cleanly on first Ctrl-C

#### Actual behavior
Launch hangs around "forever".

wjwwood, 2018-06-21:
Since the signal handler of launch is running and the shutdown event goes out and gets processed, I think it must be different from https://github.com/ros2/launch/issues/80, but I'll have to try and reproduce it here to debug it.

@clalancette what's the `python -V`?

wjwwood, 2018-06-21:
Nvm, it's bionic. I was thinking it might be Debian or something.

clalancette, 2018-06-21:
Yeah, it's bionic.  Just for reference, `python3 -V` gives me: 3.6.5

wjwwood, 2018-06-22:
I've been able to figure out that it's related to rclpy. If I don't use the ROS part of launch it works fine. Still investigating though.

wjwwood, 2018-06-29:
This is also happening on x86 Linux now as well (though perhaps not always).

mikaelarguedas, 2018-06-29:
Happens to me consistently on my bionic amd64 testing

nuclearsandwich, 2018-07-16:
I've only started looking into this on a native arm64 host. It's a host where I had previously installed .deb packages and I wasn't able to reproduce the issue with debs. What occurred instead after the first ^C was:
```
^C[WARNING] [launch.LaunchService]: user interrupted with ctrl-c (SIGINT)
root@ros2-bouncy-testing:~#signal_handler(2)
signal_handler(2)
[INFO] [launch]: process[talker-1]: process has finished cleanly
[INFO] [launch]: process[listener-2]: process has finished cleanly
```

And pressing enter, sending a clear, or any other input re-printed the command prompt. However, after downloading and running the fat archive a few times on the same host I now get the issue behavior using launch from either the debs or the fat archive. Haven't yet determined if this is some kind of bizarre contamination or just bad luck with a race condition. I straced launch to see what's going on when the first ^C is sent and it looks like it's getting stuck in  a futex

```
--- SIGINT {si_signo=SIGINT, si_code=SI_KERNEL} ---
futex(0xffff78003ef8, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0xffff78004a50, FUTEX_WAKE_PRIVATE, 1) = 1
getpid()                                = 53958
write(5, "\2", 1)                       = 1
rt_sigreturn({mask=[PIPE]})             = -1 EINTR (Interrupted system call)
rt_sigaction(SIGINT, {sa_handler=SIG_IGN, sa_mask=[], sa_flags=0}, {sa_handler=0xffff8a5e8e20, sa_mask=[INT], sa_flags=SA_RESTART}, 8) = 0
getpid()                                = 53958
write(2, "[WARNING] [launch.LaunchService]"..., 72) = 72
getpid()                                = 53958
rt_sigaction(SIGINT, {sa_handler=0x446ef8, sa_mask=[], sa_flags=0}, {sa_handler=SIG_IGN, sa_mask=[], sa_flags=0}, 8) = 0
epoll_pwait(3, [{EPOLLIN, {u32=19, u64=19}}, {EPOLLIN, {u32=21, u64=21}}, {EPOLLIN, {u32=4, u64=4}}], 5, -1000, NULL, 8) = 3
read(19, "signal_handler(2)\n", 262144) = 18
read(21, "signal_handler(2)\n", 262144) = 18
recvfrom(4, "\2", 4096, 0, NULL, NULL)  = 1
recvfrom(4, 0x38ba0e40, 4096, 0, NULL, NULL) = -1 EAGAIN (Resource temporarily unavailable)
epoll_pwait(3, [], 5, 0, NULL, 8)       = 0
futex(0xffff78000e70, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 0, NULL, 0xffffffff) = ? ERESTARTSYS (To be restarted if SA_RESTART is set)
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=54066, si_uid=0, si_status=0, si_utime=8, si_stime=6} ---
getpid()                                = 53958
write(5, "\21", 1)                      = 1
rt_sigreturn({mask=[PIPE]})             = 281472695012976
futex(0xffff78000e70, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 0, NULL, 0xffffffff) = ? ERESTARTSYS (To be restarted if SA_RESTART is set)
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=54067, si_uid=0, si_status=0, si_utime=8, si_stime=6} ---
getpid()                                = 53958
write(5, "\21", 1)                      = 1
rt_sigreturn({mask=[PIPE]})             = 281472695012976
futex(0xffff78000e70, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 0, NULL, 0xffffffff) = 0
epoll_pwait(3, [{EPOLLHUP, {u32=19, u64=19}}, {EPOLLIN|EPOLLHUP, {u32=18, u64=18}}, {EPOLLIN, {u32=4, u64=4}}, {EPOLLHUP, {u32=21, u64=21}}, {EPOLLIN|EPO
LLHUP, {u32=20, u64=20}}], 5, 0, NULL, 8) = 5
read(19, "", 262144)                    = 0
epoll_ctl(3, EPOLL_CTL_DEL, 19, 0xffffc68965c8) = 0
epoll_ctl(3, EPOLL_CTL_DEL, 18, 0xffffc6896318) = 0
recvfrom(4, "\21\21", 4096, 0, NULL, NULL) = 2
sendto(5, "\0", 1, 0, NULL, 0)          = 1
sendto(5, "\0", 1, 0, NULL, 0)          = 1
recvfrom(4, "\0\0", 4096, 0, NULL, NULL) = 2
recvfrom(4, 0x38ba17c0, 4096, 0, NULL, NULL) = -1 EAGAIN (Resource temporarily unavailable)
read(21, "", 262144)                    = 0
epoll_ctl(3, EPOLL_CTL_DEL, 21, 0xffffc68965c8) = 0
epoll_ctl(3, EPOLL_CTL_DEL, 20, 0xffffc6896318) = 0
epoll_pwait(3, [], 1, 0, NULL, 8)       = 0
write(1, "signal_handler(2)\n", 18)     = 18
close(19)                               = 0
close(18)                               = 0
wait4(54066, [{WIFEXITED(s) && WEXITSTATUS(s) == 0}], WNOHANG, NULL) = 54066
sendto(5, "\0", 1, 0, NULL, 0)          = 1
wait4(54067, [{WIFEXITED(s) && WEXITSTATUS(s) == 0}], WNOHANG, NULL) = 54067
sendto(5, "\0", 1, 0, NULL, 0)          = 1
close(21)                               = 0
close(20)                               = 0
epoll_pwait(3, [{EPOLLIN, {u32=4, u64=4}}], 1, 0, NULL, 8) = 1
recvfrom(4, "\0\0", 4096, 0, NULL, NULL) = 2
recvfrom(4, 0x38ba19f0, 4096, 0, NULL, NULL) = -1 EAGAIN (Resource temporarily unavailable)
epoll_pwait(3, [], 1, 0, NULL, 8)       = 0
epoll_pwait(3, [], 1, 0, NULL, 8)       = 0
write(1, "signal_handler(2)\n", 18)     = 18
getpid()                                = 53958
[... 38 lines cut ...]
#1  do_futex_wait (sem=sem@entry=0xffffa0000e70, abstime=0x0) at sem_waitcommon.c:111
#2  0x0000ffffb2ba2ae8 in __new_sem_wait_slow (sem=0xffffa0000e70, abstime=0x0) at sem_waitcommon.c:181
#3  0x00000000004411c0 in PyThread_acquire_lock_timed ()
#4  0x000000000059cb7c in ?? ()
#5  0x000000000079b000 in ?? ()
Backtrace stopped: not enough registers or memory available to unwind further
(gdb) info threads
  Id   Target Id         Frame
* 1    Thread 0xffffb2c05010 (LWP 41430) "python3" 0x0000ffffb2ba29c8 in futex_abstimed_wait_cancelable (private=0, abstime=0x0, expected=0,
    futex_word=0xffffa0000e70) at ../sysdeps/unix/sysv/linux/futex-internal.h:205
[... cut ...]
    futex_word=0xffffb08df738 <eprosima::fastrtps::rtps::AsyncWriterThread::cv_+40>) at ../sysdeps/unix/sysv/linux/futex-internal.h:88
  7    Thread 0xffffadd001f0 (LWP 41438) "python3" 0x0000ffffb2ba05d4 in futex_abstimed_wait_cancelable (private=<optimized out>,
    abstime=0xffffadcfe428, expected=0, futex_word=0xffffa0003ef8) at ../sysdeps/unix/sysv/linux/futex-internal.h:205
```

Still to do are to check the same issue from source (and see if I can get more info out of that backtrace as a result)

wjwwood, 2018-07-23:
Thanks @nuclearsandwich for taking some time to look into it. This information might be useful to me, so thanks for writing it up.

My current state on this is that this line:

https://github.com/ros2/launch/blob/b6085818b723944dbf8f49eae75251fe30b0ab34/launch_ros/launch_ros/default_launch_description.py#L88

Never returns after ctrl-c is done (the second ctrl-c seems to wake it up). Which is very strange because you'd think that it would eventually return after the timeout period expired (even if it did not return immediately after ctrl-c).

This odd behavior might be related with where @nuclearsandwich ended up, that a thread is getting dead locked or something within Fast-RTPS (might be affecting the `rmw_wait` method eventually).

wjwwood, 2018-07-24:
I cannot reproduce this on my 16.04 VM, I'm going to try my 18.04 VM to try and corroborate with @mikaelarguedas's comment https://github.com/ros2/launch/issues/89#issuecomment-401488395. Otherwise, I'll have to drop back into packet.net aarch64 machines.

wjwwood, 2018-08-07:
So, I've narrowed down the issue so that it only happens when:

- on bionic, and
- when using FastRTPS, and
- when using the rclpy part of launch.

So, on xenial it seems to never hang (though there is some unconfirmed evidence that it might depending on timing), and it never hangs with Connext that I've seen (or OpenSplice, though I haven't tried that myself), and it never hangs if you comment out the rclpy part of `ros2 launch` temporarily. Architecture also doesn't seem to be an issue.

I've been trying to recreate this without launch, in a script that just uses rclpy, threads, and Python signal handlers, but so far I've not been successful. I'm going to keep working on that, but at this point I'm not sure of the root cause and still not sure how to address it.

PR #124 (2018-08-17): destroy node on shutdown to avoid hang
fixes #89 

At least it fixes the issue on bionic x86_64. I didn't try aarch64 yet, but I believe it's the same issue and so the same fix. If anyone has easy access to aarch64 and could test it, that would be great.

I cannot fully explain the reason for why this is necessary or why it results in the hang. There's more work to be done here (in my opinion) so that this doesn't happen to typical users. There's no reason that I should have to manually destroy the node.

Thanks to @nuclearsandwich for giving me the last hint I needed to find this workaround.  [... cut]
closed 2018-08-17 by wjwwood