ros2/launch #95: 'ros2 launch' invalid choice

shiveshkhaitan, 2018-07-02:
I do not get the option of launch after typing ros2
This is the error msg 
invalid choice: 'launch' (choose from 'daemon', 'extension_points', 'extensions', 'lifecycle', 'msg', 'node', 'param', 'pkg', 'run', 'security', 'service', 'srv', 'topic')

sloretz, 2018-07-02:
Would you mind filling out the issue template? It sounds like `ros2launch` might not have been built, but there isn't enough information to say why.

```
## Bug report

**Required Info:**

- Operating System:
  - <!-- OS and version (e.g. Windows 10, Ubuntu 16.04...) -->
- Installation type:
  - <!-- binaries or from source  -->
- Version or commit hash:
  - <!-- Output of git rev-parse HEAD, release version, or repos file  -->
- DDS implementation:
  - <!-- rmw_implementation used (e.g. Fast-RTPS, RTI Connext, etc -->
- Client library (if applicable):
  - <!-- e.g. rclcpp, rclpy, or N/A -->

#### Steps to reproduce issue
<!-- Detailed instructions on how to reliably reproduce this issue http://sscce.org/
``` code that can be copy-pasted is preferred ``` -->

#### Expected behavior

#### Actual behavior

#### Additional information
```

wjwwood, 2018-07-11:
I'm going to close this until more information is provided. Feel free to update your issue and comment here, at which point I'll reopen this issue.

Also you can ask questions like this on https://answers.ros.org.

closed 2018-07-11 by wjwwood

oscar-lima, 2018-08-03:
I have the same problem, here is the bug report:

## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - binaries
- Version or commit hash:
  - ardent
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
```ros2``` tab, tab!

#### Expected behavior
launch daemon extension_points extensions msg node pkg run security service srv topic

#### Actual behavior
daemon extension_points extensions msg node pkg run security service srv topic

#### Additional information
roscli problem?, if I force the execution like: ```ros2 launch```, I get:

```
usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...
ros2: error: argument Call `ros2 <command> -h` for more detailed usage.: invalid choice: 'launch' (choose from 'daemon', 'extension_points', 'extensions', 'msg', 'node', 'pkg', 'run', 'security', 'service', 'srv', 'topic')
```

sloretz, 2018-08-03:
Hi @oscar-lima, thanks for filling out the template. The `ros2 launch` command is new in Bouncy. That behavior is expected in the Ardent release.

oscar-lima, 2018-08-03:
Ok, Thanks! @sloretz

felixf4xu, 2023-05-20:
I'm building ros2 from source and I had the same issue when running `ros2 run ...`

In my case, I clean the folder and rebuild `ros2run` and it's fixed.