ros2/launch #98: `ros2 launch` should autocomplete only packages that have launch files

mikaelarguedas, 2018-07-04:
Currently `ros2 launch` autocompletes with the list of all packages.

We could provide a way to register launchfiles to the index in CMake and use that resource to autocomplete only packages that have launchfiles, and only files that are launch files.

dirk-thomas, 2018-07-05:
> We could provide a way to register launchfiles to the index in CMake

How about Python packages? Registering them in the index will require quite some custom logic in the `setup.py` which is probably not feasible. So relying on the index might not be an option.

wjwwood, 2018-07-09:
> How about Python packages? Registering them in the index will require quite some custom logic in the setup.py which is probably not feasible. So relying on the index might not be an option.

That's already the case for all python packages which are ROS packages. I don't think that's a reason to avoid using the index. If it's inconvenient, then we should make it more convenient, not avoid using it despite the fact that it's the most efficient way to handle this.

Crawling for launch files in the share directory of all packages found is the only alternative I'm aware of that doesn't require information stored at install time and that's not going to be fast enough in my opinion.

dirk-thomas, 2018-07-09:
> If it's inconvenient, then we should make it more convenient

Do you have a proposal how to achieve that with the constraint that you can't rely on any Python module to be available in a `setup.py` file?

dirk-thomas, 2018-07-09:
I am not trying to "stall" this work and I don't have a proposal for a fast lookup without the index.

> We could reevaluate our decision to not import things in setup.py, continue duplicating code for ament index installation including for having launch files, or find some other solution.

We can't just import other modules since they are not guaranteed to be available. Therefore we are currently duplicating logic like this (e.g. to register a package at the index). As far as I know it is necessary.

I was asking if you have a proposal for "find some other solution" since to my knowledge using a module doesn't reliably work since there is no way to express that dependency and duplication is not desired.

wjwwood, 2018-07-09:
> Registering them in the index will require quite some custom logic in the setup.py which is probably not feasible.

I saw the above as "we shouldn't do this", that's what "not feasible" means to me.

I don't have a suggestion for either how to avoid the custom logic in the `setup.py`'s (the infeasible part) or how to do the lookup effectively without the ament index.

So I think we either have to not have this feature request fulfilled or we have to have custom logic in `setup.py` files. My preference is for the latter, you seemed to prefer the former (which is why I think it's stalling this feature from being implemented).