ros2/launch #103: [launch] refactor introspection system

wjwwood, 2018-07-11:
Some effort was already invested to express intent without side-effects in the new launch api, and each entity in a launch description can express itself and related, or "sub" entities. However, the introspection interfaces of various entities (Actions, Event Handlers, Events, Subsitutions, etc...) need to be refined to avoid impassible and opaque boundaries in the description when introspecting.

I have some ideas of what needs to change already, but ran out of time before the bouncy release. The `launch.LaunchDescriptionEntity` interfaces for this are the most "mature" and reflect somewhat closely what I'd like to update the other entities (like `EventHandler` and `Substitution`) to look like in the future:

https://github.com/ros2/launch/blob/b78d87f31bb995c7226df66826e1a57079ae3a3f/launch/launch/launch_description_entity.py#L57-L101

The above interfaces may also need some work. Additionally, I think that right now the mechanism to "render" an entity during introspection (render to text for printing on the console in the case of `launch.LaunchIntrospector`) needs to be placed with or along side the entities themselves. Currently this logic is in the `LaunchIntrospector`, which was only done to get something working rapidly:

https://github.com/ros2/launch/blob/b78d87f31bb995c7226df66826e1a57079ae3a3f/launch/launch/launch_introspector.py#L72-L140

Lastly, specific entities need to implement these interfaces and handle any special logic to make them useful during introspection.

rolker, 2025-02-10:
I'm interested in helping move this along. One question I have is can `__repr__()` or `__str__()` be used instead of `describe()`?

wjwwood, 2025-02-28:
@rolker great to hear, let me know if I can help in other ways.

As for your question:

> One question I have is can __repr__() or __str__() be used instead of describe()?

I think we could have `__str__()` as a replacement for `describe()`. `__repr__` is more like "engineering data" to me, in that it's goal is to be unambiguous rather than nice to look at, per se.

I would still keep `describe()` in the (perhaps rare case) that the author of an `Action` would like to do something different for introspection of launch actions than for simply printing them with `__str__()`. Having a default implementation of `describe()` that calls `__str__()` would be fine I think.