ros2/launch #117: [launch_ros] add ability to pass parameters to Node actions

wjwwood, 2018-07-12:
Currently you can pass remapping arguments to Node as a `dict`, but to set parameters in a node from a launch file you need to pass a normal argument which points to an existing `.yaml` file.

To pass parameters via the launch file (without predefining them in an external `.yaml` file) will ultimately require the ability to pass parameters via the command line one at a time, which as of right now is not implemented yet.

In the meantime, individually specified parameters in a launch file for a node could be written into a temporary `.yaml` file by `launch` and then passed to the node, deleting the file after the node exits. However, I think this would step on a user specified parameter `.yaml` file unless the node can take more than one right now (it was discussed, but I don't know off-hand if it was implemented). In either case, if launch also lets you specify one or more `.yaml` files to be used as parameters (in addition to, and at the same time as, individually specified parameters for a node), then launch could just combine all of the yaml files and parameter kay-value pairs into a single `.yaml` file, cascading them in order (overwriting keys that appear more than once, with last wins).

It might look something like this:

```python
launch_ros.actions.Node(...,
  parameters=[
    '/path/to/first_parameter_file.yaml',
    launch_ros.actions.node.NodeParameters({
      'foo': '1',
      // can have substitutions
      [LaunchConfiguration('ros_namespace'), '.my_node.executable_location']:
        [ExecutableInPackage('my_node', 'my_package')],
      ...,
    },
    [PackageShareDir('my_package'), Dir('config') / 'my_config_file.yaml'],
  ],
)
```

sloretz, 2018-07-12:
> In the meantime, individually specified parameters in a launch file for a node could be written into a temporary .yaml file by launch and then passed to the node, deleting the file after the node [...]

Multiple yaml files can be given on the command line. If two files specify a value for the same parameter then the value from the later file is used.

```
ros2 run demo_nodes_cpp talker __params:=generic-default-values.yaml __params:=more-specific-values.yaml
```

wjwwood, 2018-07-12:
Cool thanks!

sloretz, 2018-09-14:
Is this ticket complete now that #135 has been merged?

dhood, 2018-09-14:
nah, it didn't include passing parameters as dicts

dhood, 2018-09-14:
(which I'm working on now)

closed 2018-09-21 by dhood

poonam1120, 2018-11-23:
Hi, 
I want to set value of use_sim_time as true when I am launching node via .launch.py file but I am not able to do this.
Node is getting launched with yaml file but not sure how to set value for  use_sim_time here.
/**my  code snippet**/
launch_ros.actions.Node(
            package='test', node_executable='test_node', node_name='test_node',
	    output='screen',
	    parameters=[
                parameters_file_path,
                str('/home/root-pc/test.yaml'),
                [EnvironmentVariable(name='FILE_PATH'), os.sep, 'test.yaml'],
           ],
           ),

****/
Any help would be appreciated.

Thanks.

sloretz, 2018-11-26:
Hi @poonam1120,

I think this question would be a better fit for https://answers.ros.org/questions/ . Many more people watch that site, so you're likely to get an answer quicker. Comments on closed issues often don't get responses because closed issues on github aren't very visible.