ros2/launch #139: Don't require node name to be specified if parameter dictionary passed to Node action

dhood, 2018-09-15:
The node namespace is always specified when launching a Node action (defaults to `/`), but the node name is optional.

When parameters are passed to a Node action as a dictionary, they are written to a yaml file. Yaml files of parameters currently require the node name to be specified.

Because of this, if a dictionary of parameters is passed to a Node action, there is a requirement that the node name is also specified.

This requirement can be removed if parameters are passed individually on the command line to the node in a way that doesn't require the node's name (feature not implemented in rcl yet), or if yaml files are made to no longer require the node name.

wjwwood, 2018-09-15:
> This requirement can be removed if parameters are passed individually on the command line to the node in a way that doesn't require the node's name (feature not implemented in rcl yet), or if yaml files are made to no longer require the node name.

Is this something you think will be possible?

If we can ever pass arguments over the command line I was assuming that we'd need to specify the node name as well. Put another way, anything you could pass over the command line should also be representable in the yaml file as well. So if you have "node-agnostic" parameters on the command line you could do the same in the yaml file. Or if you have parameters that only work if there's a single node in the executable (enforced somehow) then you could have an equivalent magic section in the yaml file.

I don't think this (requiring that the node name is given when passing parameters too) will be a requirement that can be dropped ever unfortunately...

For context, Lifecycle nodes require the node name and so does roslaunch from ROS 1 (in all cases).

dhood, 2018-09-17:
Not requiring the node name to be passed one day seems plausible to me. This is my reasoning:

1. Node agnostic parameters in yaml files.
    1. Parsed parameters are stored in a FQN node name - to - parameters mapping. We could have an empty name or some other invalid/reserved node name in the storage reflect that the parameters should apply to all nodes. The parser would need to be updated to not require a node name before the `ros__parameters` key.
1. Node agnostic parameters over the command line.
    1. Remapping args can already be passed either to a specific node (`talker:chatter:=new_chatter`) or to all nodes in a process (`chatter:=new_chatter`). Figured we'd just implement the equivalent for parameters.

> I don't think this (requiring that the node name is given when passing parameters too) will be a requirement that can be dropped ever unfortunately...

If requiring that the node name be specified was a reasonable thing for `roslaunch` in ROS 1 / Lifecycle node actions, then we might be fine to just make specifying the node name always a requirement for the general Node actions. But, it doesn't seem to me that there is a reason (yet) that we *must* do it in the the long term (in the short term, yes, because the above isn't implemented). (Perhaps you have additional context around why it is required long-term because of the component launching you're working on, but I didn't pick up on it from your previous comment).