ros2/launch #144: [launch] Using TimerAction results in exception

crdelsey, 2018-10-03:
**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - source
- Version or commit hash:
  - master/HEAD as of 3 Oct 2018 8597466f
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - 

#### Steps to reproduce issue
`ros2 launch` the launch file below
```python
import os
from launch import LaunchDescription
import launch.actions
import launch_ros.actions

def generate_launch_description():
    return LaunchDescription([
        launch.actions.TimerAction(
            actions = [
                launch_ros.actions.Node( package='demo_nodes_cpp', node_executable='listener', output='screen')
                ], period = 5.0),
    ])
```

#### Expected behavior
The demo_nodes listener node gets launched
#### Actual behavior
`[ERROR] [launch.LaunchService]: Caught exception in launch (see debug for traceback): 'tuple' object has no attribute 'describe_sub_entities'`

#### Additional information
The full backtrace is here
```
[DEBUG] [launch]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
[DEBUG] [launch]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
[DEBUG] [launch.LaunchService]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f239e069ba8>'
[DEBUG] [launch.LaunchService]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f239e069ba8>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x7f239e069908>'
[DEBUG] [launch.LaunchService]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f239e069cc0>'
[DEBUG] [launch.LaunchService]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f239e069cc0>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x7f239e069908>'
[DEBUG] [launch.LaunchService]: Traceback (most recent call last):
  File "/home/crdelsey/src/ros_dev_workspace/navstack_dependencies_ws/build/launch/launch/launch_service.py", line 340, in run
    self.__loop_from_run_thread.run_until_complete(run_loop_task)
  File "/usr/lib/python3.5/asyncio/base_events.py", line 387, in run_until_complete
    return future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 274, in result
    raise self._exception
  File "/usr/lib/python3.5/asyncio/tasks.py", line 241, in _step
    result = coro.throw(exc)
  File "/home/crdelsey/src/ros_dev_workspace/navstack_dependencies_ws/build/launch/launch/launch_service.py", line 236, in __run_loop
    await process_one_event_task
  File "/usr/lib/python3.5/asyncio/futures.py", line 361, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/usr/lib/python3.5/asyncio/tasks.py", line 296, in _wakeup
    future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 274, in result
    raise self._exception
  File "/usr/lib/python3.5/asyncio/tasks.py", line 239, in _step
    result = coro.send(None)
  File "/usr/lib/python3.5/asyncio/coroutines.py", line 121, in send
    return self.gen.send(value)
  File "/home/crdelsey/src/ros_dev_workspace/navstack_dependencies_ws/build/launch/launch/launch_service.py", line 179, in _process_one_event
    await self.__process_event(next_event)
  File "/usr/lib/python3.5/asyncio/coroutines.py", line 105, in __next__
    return self.gen.send(None)
  File "/home/crdelsey/src/ros_dev_workspace/navstack_dependencies_ws/build/launch/launch/launch_service.py", line 198, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/home/crdelsey/src/ros_dev_workspace/navstack_dependencies_ws/build/launch/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/home/crdelsey/src/ros_dev_workspace/navstack_dependencies_ws/build/launch/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/home/crdelsey/src/ros_dev_workspace/navstack_dependencies_ws/build/launch/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/home/crdelsey/src/ros_dev_workspace/navstack_dependencies_ws/build/launch/launch/actions/include_launch_description.py", line 113, in visit
    declared_launch_arguments = launch_description.get_launch_arguments()
  File "/home/crdelsey/src/ros_dev_workspace/navstack_dependencies_ws/build/launch/launch/launch_description.py", line 105, in get_launch_arguments
    process_entities(self.entities, _conditional_inclusion=conditional_inclusion)
  File "/home/crdelsey/src/ros_dev_workspace/navstack_dependencies_ws/build/launch/launch/launch_description.py", line 103, in process_entities
    entity.describe_conditional_sub_entities(), _conditional_inclusion=True)
  File "/home/crdelsey/src/ros_dev_workspace/navstack_dependencies_ws/build/launch/launch/launch_description.py", line 101, in process_entities
    entity.describe_sub_entities(), _conditional_inclusion=False)
AttributeError: 'tuple' object has no attribute 'describe_sub_entities'

[ERROR] [launch.LaunchService]: Caught exception in launch (see debug for traceback): 'tuple' object has no attribute 'describe_sub_entities'
[DEBUG] [launch]: emitting event: 'launch.events.Shutdown'
[DEBUG] [launch.LaunchService]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f23a495ccc0>'
[DEBUG] [launch.LaunchService]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f23a495ccc0>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7f239e0690b8>'
[DEBUG] [launch.LaunchService]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f23a495ccc0>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7f23a16ca390>'
[WARNING] [asyncio]: Executing %s took %.3f seconds
```

Also, commenting out the `describe_conditional_sub_entities` method of the `TimerAction` class seems to allow the launch file to work as intended.

jacobperron, 2018-10-04:
Looks like this recursive call in `launch_description.py` is not implemented properly:

https://github.com/ros2/launch/blob/8597466f5da210fecff4c70e3db0f9935ccd7390/launch/launch/launch_description.py#L103

`describe_conditional_sub_entities()` returns a list of tuples where each tuple has a description of the condition and a list of sub-entities. Perhaps, one solution would be to first check if the entity is a `tuple` (implying that it is conditional) and then passing in the second element of the tuple to the recursive call. Maybe not a great hack...

e.g.

```python
# assume tuple contains a list of entities at index 1
if isinstance(entity, tuple)
    entity = entity[1]

process_entities(
    entity.describe_sub_entities(), _conditional_inclusion=False)
process_entities(
    entity.describe_conditional_sub_entities(), _conditional_inclusion=True)
```

We could be more robust and add additional checks like so:

```python
if isinstance(entity, list) or (isinstance(entity, tuple) and isinstance(entity[1], list)):
    process_entities(
        entity.describe_sub_entities(), _conditional_inclusion=False)
    process_entities(
        entity.describe_conditional_sub_entities(), _conditional_inclusion=True)
```

jacobperron, 2018-10-04:
We could also make use of the `_condition_inclusion` argument in the check for a hint to if we expect a tuple.

rotu, 2019-05-29:
Still broken :-1:

pbaughman, 2019-06-06:
It's surprising to me that [the tests for the timer action](https://github.com/ros2/launch/blob/master/launch/test/launch/test_timer_action.py) don't catch this issue.  I wonder if it only happens when using the ros2launch CLI tool. . . anyway, this is going to affect me too very soon, so I'm going to dive in.

pbaughman, 2019-06-06:
Oh nevermind - I see - it's a result of the launch description being part of another 'include launch description' action and calling 'get_launch_arguments' which never happens in the tests

pbaughman, 2019-06-06:
@jacobperron  Can't we do something like:

```
. . .
                else:                                                                        
                    from .actions import IncludeLaunchDescription                            
                    if isinstance(entity, IncludeLaunchDescription):                         
                        # Do not check launch arguments of included descriptions.            
                        continue                                                             
                    process_entities(                                                        
                        entity.describe_sub_entities(), _conditional_inclusion=False)        
                    for conditional_sub_entity in entity.describe_conditional_sub_entities():
                        process_entities(                                                    
                            conditional_sub_entity[1],                                       
                            _conditional_inclusion=True)                                     
```

We don't need any type checking - `describe_conditional_sub_entities` always returns a list of (description, List[LaunchDescriptionEntity]) tuples and `process_entities` always takes a list of LaunchDescriptionEntity objects.  We just need to convert the former into the latter before we recurse.

Am I missing something?

jacobperron, 2019-06-06:
@pbaughman It's been a while since I looked at this, but based on the method signatures that seems correct to me.

PR #257 (2019-06-07): Fix get_launch_arguments to not crash on conditional sub entities
  - This resolves #144 where the bug was being triggered by a TimerAction
  - Test case added to test_launch_description.py
commit 65c194ee (2019-06-07): Fix get_launch_arguments to not crash on conditional sub entities (#257)

- This resolves #144 where the bug was being triggered by a TimerAction
  - Test case added to test_launch_description.py
commit fa1dd13b (2019-06-12): Fix get_launch_arguments to not crash on conditional sub entities (#257)

- This resolves #144 where the bug was being triggered by a TimerAction
  - Test case added to test_launch_description.py
closed 2019-06-07 by jacobperron