ros2/launch #157: ValueError raised using --print on an ASCII launch file

sloretz, 2018-12-11:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu bionic
- Installation type:
  - debs
- Version or commit hash:
  - crystal pre-release
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
Try to `--print` the installed `talker_listener.launch.py`
```
ros2 launch --print /opt/ros/crystal/share/demo_nodes_cpp/launch/topics/talker_listener.launch.py
```

#### Expected behavior

The content of `talker_listener.launch.py` would be printed to the console.

#### Actual behavior

Raises value error when using `--print`
```bash
$ ros2 launch --print /opt/ros/crystal/share/demo_nodes_cpp/launch/topics/talker_listener.launch.py 

Notice: ValueError can occur if your launch file ('/opt/ros/crystal/share/demo_nodes_cpp/launch/topics/talker_listener.launch.py') is a binary file and not a Python file.

ValueError: 'ascii' codec can't encode characters in position 71-73: ordinal not in range(128)
```

Outputting file on the console shows it is a text file.

```bash
$ cat /opt/ros/crystal/share/demo_nodes_cpp/launch/topics/talker_listener.launch.py
# Copyright 2018 Open Source Robotics Foundation, Inc.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

"""Launch a talker and a listener."""

from launch import LaunchDescription
import launch_ros.actions

def generate_launch_description():
    return LaunchDescription([
        launch_ros.actions.Node(
            package='demo_nodes_cpp', node_executable='talker', output='screen'),
        launch_ros.actions.Node(
            package='demo_nodes_cpp', node_executable='listener', output='screen'),
    ])
```

`file` agrees that it is an ASCII file

```bash
$ file /opt/ros/crystal/share/demo_nodes_cpp/launch/topics/talker_listener.launch.py
/opt/ros/crystal/share/demo_nodes_cpp/launch/topics/talker_listener.launch.py: Python script, ASCII text executable
```

#### Additional information

It's not just installed python files. I copy/pasted another launch file into a new file in the hope directory and the same `ValueError` was raised.

sloretz, 2018-12-11:
>The command works for me outputting:

On bionic? Using crystal debs or from source?

dirk-thomas, 2018-12-11:
> On bionic? Using crystal debs or from source?

Yes, using Debian packages on Bionic - same as in the issue template above.

closed 2018-12-11 by sloretz