ros2/launch #159: Remove legacy API

sloretz, 2019-01-08:
The legacy API should be removed.

<details><summary>Places where legacy API is used</summary>

```
./ros2/system_tests/test_cli_remapping/test/test_cli_remapping.py:from launch.legacy import LaunchDescriptor
./ros2/system_tests/test_cli_remapping/test/test_cli_remapping.py:from launch.legacy.exit_handler import primary_exit_handler
./ros2/system_tests/test_cli_remapping/test/test_cli_remapping.py:from launch.legacy.launcher import DefaultLauncher
./ros2/rcutils/test/test_logging_long_messages.py:from launch.legacy import LaunchDescriptor
./ros2/rcutils/test/test_logging_long_messages.py:from launch.legacy.exit_handler import ignore_exit_handler
./ros2/rcutils/test/test_logging_long_messages.py:from launch.legacy.launcher import DefaultLauncher
./ros2/rcutils/test/test_logging_long_messages.py:from launch.legacy.output_handler import ConsoleOutput
./ros2/rcutils/test/test_logging_output_format.py:from launch.legacy import LaunchDescriptor
./ros2/rcutils/test/test_logging_output_format.py:from launch.legacy.exit_handler import ignore_exit_handler
./ros2/rcutils/test/test_logging_output_format.py:from launch.legacy.launcher import DefaultLauncher
./ros2/rcutils/test/test_logging_output_format.py:from launch.legacy.output_handler import ConsoleOutput
./ros2/launch/launch_testing/launch_testing/legacy/__init__.py:from launch.legacy.output_handler import LineOutput
./ros2/launch/launch_testing/launch_testing/__init__.py:from launch_testing.legacy import create_handler  # noqa: F401
./ros2/launch/launch_testing/launch_testing/__init__.py:from launch_testing.legacy import get_default_filtered_patterns  # noqa: F401
./ros2/launch/launch_testing/launch_testing/__init__.py:from launch_testing.legacy import get_default_filtered_prefixes  # noqa: F401
./ros2/launch/launch_testing/launch_testing/__init__.py:from launch_testing.legacy import get_rmw_output_filter  # noqa: F401
./ros2/launch/launch_testing/launch_testing/__init__.py:from launch_testing.legacy import InMemoryHandler  # noqa: F401
./ros2/launch/launch_testing/launch_testing/__init__.py:from launch_testing.legacy import UnmatchedOutputError  # noqa: F401
./ros2/launch/launch_testing/test/test_env_passing.py:from launch.legacy import LaunchDescriptor
./ros2/launch/launch_testing/test/test_env_passing.py:from launch.legacy.exit_handler import primary_exit_handler
./ros2/launch/launch_testing/test/test_env_passing.py:from launch.legacy.launcher import DefaultLauncher
./ros2/launch/launch_testing/test/test_matching.py:from launch.legacy import LaunchDescriptor
./ros2/launch/launch_testing/test/test_matching.py:from launch.legacy.exit_handler import ignore_exit_handler
./ros2/launch/launch_testing/test/test_matching.py:from launch.legacy.launcher import DefaultLauncher
./ros2/launch/launch/setup.py:            'launch = launch.legacy.main:main',
./ros2/launch/launch/test/legacy/test_non_primary_return_code.py:from launch.legacy import LaunchDescriptor
./ros2/launch/launch/test/legacy/test_non_primary_return_code.py:from launch.legacy.exit_handler import primary_exit_handler
./ros2/launch/launch/test/legacy/test_non_primary_return_code.py:from launch.legacy.launcher import DefaultLauncher
./ros2/launch/launch/test/legacy/test_interrupt_default_launcher.py:from launch.legacy import LaunchDescriptor
./ros2/launch/launch/test/legacy/test_interrupt_default_launcher.py:from launch.legacy.launcher import DefaultLauncher
./ros2/launch/launch/test/legacy/test_multiple_launch.py:from launch.legacy import LaunchDescriptor
./ros2/launch/launch/test/legacy/test_multiple_launch.py:from launch.legacy.launcher import DefaultLauncher
./ros2/launch/launch/test/legacy/launch_counter.py:from launch.legacy.exit_handler import ignore_exit_handler
./ros2/launch/launch/test/legacy/launch_counter.py:from launch.legacy.output_handler import FileOutput
./ros2/launch/launch/test/legacy/test_launch_args.py:from launch.legacy.arguments import get_launch_args
./ros2/launch/launch/test/legacy/launch_multiple_counters.py:from launch.legacy import LaunchDescriptor
./ros2/launch/launch/test/legacy/launch_multiple_counters.py:from launch.legacy.exit_handler import ignore_exit_handler
./ros2/launch/launch/test/legacy/launch_multiple_counters.py:from launch.legacy.exit_handler import restart_exit_handler
./ros2/launch/launch/test/legacy/launch_multiple_counters.py:from launch.legacy.loader import load_launch_file
./ros2/launch/launch/test/legacy/launch_multiple_counters.py:from launch.legacy.output_handler import ConsoleOutput
./ros2/launch/launch/test/legacy/launch_multiple_counters.py:from launch.legacy.output_handler import FileOutput
./ros2/launch/launch/test/legacy/test_launch_with_coroutine.py:from launch.legacy import LaunchDescriptor
./ros2/launch/launch/test/legacy/test_launch_with_coroutine.py:from launch.legacy.exit_handler import primary_exit_handler
./ros2/launch/launch/test/legacy/test_launch_with_coroutine.py:from launch.legacy.launcher import DefaultLauncher
./ros2/launch/launch/test/legacy/test_launch_with_coroutine.py:from launch.legacy.loader import load_launch_file
./ros2/launch/launch/launch/legacy/launcher.py:from launch.legacy.exit_handler import ExitHandlerContext
./ros2/launch/launch/launch/legacy/launcher.py:from launch.legacy.launch import LaunchState
./ros2/launch/launch/launch/legacy/launcher.py:from launch.legacy.protocol import SubprocessProtocol
./ros2/launch/launch/launch/legacy/launcher.py:from launch.legacy.task import TaskState
./ros2/launch/launch/launch/legacy/main.py:from launch.legacy import LaunchDescriptor
./ros2/launch/launch/launch/legacy/main.py:from launch.legacy.launcher import DefaultLauncher
./ros2/launch/launch/launch/legacy/main.py:from launch.legacy.loader import load_launch_file
./ros2/launch/launch/launch/legacy/__init__.py:from launch.legacy.exit_handler import default_exit_handler
./ros2/launch/launch/launch/legacy/__init__.py:from launch.legacy.output_handler import CompositeOutputHandler
./ros2/launch/launch/launch/legacy/__init__.py:from launch.legacy.output_handler import ConsoleOutput
./ros2/launch/launch/launch/__init__.py:from . import legacy
./ros2/launch/launch/launch/__init__.py:    'legacy',
```
</details>
<br>

Packages to update
- [ ] test_cli_remapping
- [ ] rcutils
- [ ] launch_testing
- [ ] launch

hidmic, 2019-02-07:
Well, this ended up being much more work than expected. Above list isn't exhaustive: it's missing all the `*.py.in`  files that make use of the legacy launch API.

closed 2019-02-28 by mjcarroll

hidmic, 2019-02-28:
Do we want to remove any trace of the legacy API too? @mjcarroll @dirk-thomas

mjcarroll, 2019-02-28:
I think since nothing is using it, it's safe to drop.  Sorry, I thought that was part of the previous PR here, so I closed the issue.  If we need to, we can reopen to track this.

hidmic, 2019-03-11:
Alright, legacy launch API completely deprecated after https://github.com/ros2/launch/pull/191. Dashing release changelist reflects this after https://github.com/ros2/ros2_documentation/pull/135.