ros2/launch #163: XML/YAML Front-end

hidmic, 2019-01-14:
## Feature request

This is a feature request to implement a declarative markup (XML, YAML, TOML, etc.) front-end for launch descriptions. Acceptance criteria is limited to TBD.

#### Feature description

- Optionally narrow down feature set for a first iteration.
- Draft a format standardization document in https://github.com/ros2/design.
- Gather feedback from potential users (i.e. community and customers).
- Implement parser, from markup based description into `LaunchDescription` instances.
- Update `ros2launch` CLI to also understand declarative launchfiles using the parser.

#### Implementation considerations

**ros2launch**

Consider refactoring to ease extension through integration of external parsers (using Python modules?).

**launch_\[markup_fmt\]**

Weigh different markup formats. Some tentative XML can be found in https://github.com/ros2/design/pull/163. Some draft YAML can be found [here](https://gist.github.com/mjcarroll/1609c19433868a0fd863fbfd2a1506b1).

With the current launch API being already quite declarative, a 1-on-1 mapping may be possible (though this is still a conjecture).

dirk-thomas, 2019-01-14:
Also take the existing ROS 1 roslaunch XML into account: http://wiki.ros.org/roslaunch/XML

hidmic, 2019-02-18:
A first XML launch format specification can be found in ros2/design#207. Several approaches on how to implement launch front ends (in general, not specific to XML) are described in ros2/design#208. 

@wjwwood and I think that [ADP](https://github.com/ros2/design/pull/208/files#diff-1acde8639704c397705390e306dba4bfR190) offers the best tradeoff between complexity, scalability and extensibility, and thus we'd rather pursue that one. As a summary of what a first Python implementation of such a design would entail, consider the following todo list:

1. [ ] Define an abstract `launch_frontend.Entity` class. (~1 day for 1 FTE)
2. [ ] Implement (and likely refine base API of) a `launch_xml.Entity` class to wrap XML tags (as represented by any [standard xml parsing module](https://docs.python.org/3/library/xml.html)). Thoroughly document the mapping from XML schema to object abstraction. (~4 days for 1 FTE)
3. [ ] Implement a `launch_frontend.expose_action(*, aliases)` decorator for functions or `launch.Action` subclasses that have a `parse(*, entity: launch_frontend.Entity, parser: launch_frontend.Parser) -> launch.Action` class method that keeps the mapping from known actions aliases to parsing functions. (~1 day for 1 FTE)
4. [ ] Implement a `launch_frontend.expose_substitution(*, aliases)` decorator for functions or `launch.Substitution` subclasses that have a `parse(*, args: Text, parser: launch_frontend.Parser) -> launch.Substitution` class method that keeps the mapping from known substitutions aliases to parsing functions. (~1 day for 1 FTE)
5. [ ] Implement a `launch_xml.InterpolationSubstitution` class to handle substitutions embedded in text. (~2 day for 1 FTE)
6. [ ] Define an abstract `launch_frontend.Parser` class, with methods: (~1 day for 1 FTE)
    1. [ ] `parse_action(entity: launch_frontend.Entity) -> launch.Action`
    2. [ ] `parse_substitution(value: Text) -> launch.Substitution`
    3. [ ] `parse_description(stream: io.TextIOBase) -> launch.LaunchDescription`
7. [ ] Implement (and likely refine base API of) a `launch_xml.Parser` class that makes use of available registries and associated `launch_xml.Entity` to actually load some XML. (~2 day for 1 FTE)
8. [ ] Extend `launch_frontend.expose_action` and `launch_frontend.expose_substitution `decorators to inspect the wrapped class constructor annotations, if any, to provide a generic parsing function as a fallback if one's not available. Thoroughly document inference rules (~3 days for 1 FTE)
9. [ ] Implement parsing functions for all actions and substitution for which said functions cannot be reasonably inferred. (~3 days for 1 FTE)
10. [ ] Integrate `launch_xml.Parser` to launch source loading actions/functions. (~3 days for 1 FTE)

That is, **roughly** (and I want to emphasize that qualifier) three (3) weeks worth of an FTE work are required (including reviews if running at a good pace). We may be able to take some shortcuts, though I'm not sure we can radically reduce the sheer amount of man power needed. @mjcarroll @sloretz @ivanpauno thoughts?

closed 2019-07-30 by ivanpauno