ros2/launch #164: [launch] add new Action called SetEnvironmentVariable

wjwwood, 2019-01-16:
This action would just take a name and value, and it would set an environment variable when executed. The name and value should take substitutions so that this action may be used to extend environment variables as well as overwrite them, e.g. `SetEnvironmentVariable(name='PATH', value=[EnvironmentVariable('PATH'), '/some/other/bin'])`.

wjwwood, 2019-01-25:
Awesome! The best way to do this is to open a pull request (https://help.github.com/articles/about-pull-requests/) against this repository with the feature implemented. The best way to see how to do that is to probably emulate another pull request that does something similar, for example: https://github.com/ros2/launch/pull/140

wjwwood, 2019-02-09:
@manaswinidas

I haven't heard from you in a bit (which is totally fine!), but because of that I'm going to put this back into our backlog. If you're still working on it, feel free to speak up and I'll put it back in progress, or if you don't have time right now but do in the future, speak up then and I can put in progress again at the point.

Thanks!

oswinso, 2019-03-23:
I'm not sure if anyone is working on this right now, but I'd like to try and work on this issue.

PR #216 (2019-03-28): Add SetEnvironmentVariable and UnsetEnvironmentVariable actions #164
Closes #164. Also added tests for `SetEnvironmentVariable` and `UnsetEnvironmentVariable` actions.
closed 2019-03-28 by wjwwood