ros2/launch #180: Shutdown action does not interrupt timers

pbaughman, 2019-02-09:
## Bug report
A launch.actions.TimerAction will keep a launch service running even after a Shutdown action

It's possible this is expected behavior, but from the [description of a shutdown action](https://github.com/ros2/launch/blob/master/launch/launch/actions/shutdown_action.py#L29) I'd expect it to terminate everything right away
**Required Info:**

- Operating System: Ubuntu 16.04
  - 
- Installation type: source
  - 
- Version or commit hash: b665104c8dc30be144c3ac3c4c16037d6b191900
  - 
- DDS implementation:
  - 
- Client library (if applicable): rclpy
  - 

#### Steps to reproduce issue

```
import time                                                                 
import launch                                                               
import launch.actions                                                       
import launch_ros.actions                                                   
                                                                            
def generate_launch_description():                                          
                                                                            
    return launch.LaunchDescription([                                       
                                                                            
        launch_ros.actions.Node(                                            
            package="demo_nodes_py",                                        
            node_executable='talker',                                       
        ),                                                                  
                                                                            
        launch.actions.TimerAction(                                         
            period="5",                                                     
            actions=[                                                       
                launch.actions.Shutdown(reason="Five second timeout")       
            ]                                                               
        ),                                                                  
                                                                            
        launch.actions.TimerAction(                                         
            period="20",                                                    
            actions=[                                                       
                launch.actions.Shutdown(reason="Twenty second timeout")     
            ]                                                               
        )                                                                   
                                                                            
    ])                                                                                                                                                  
                                                                            
if __name__ == "__main__":                                                  
    launch_service = launch.LaunchService()                                 
    launch_service.include_launch_description(generate_launch_description())
    start = time.time()                                                     
    launch_service.run()                                                    
    end = time.time()                                                       
    print(end - start)                                                      
```

#### Expected behavior
I'd expect the above program to terminate after about 5 seconds, once the first TimerAction causes a shutdown

#### Actual behavior
The talker node is shut down after 5 seconds, but the call to launch_service.run() does not return until 20 seconds have elapsed

#### Additional information
Just let me know if my expectations are wrong here.  Since the docs say "shut down a launched system" instead of "shut down all running processes" I sort of expect the whole thing to wrap up after 5 seconds

wjwwood, 2019-02-09:
I didn't intentionally make it so that timers would hold up the system, however, that might be important to the design right now. For example, when you call for shutdown, timers are created, that need to not immediately expire, to escalate to SIGTERM and SIGKILL for processes that are misbehaving.

I think the appropriate solution is that timers are changed to intercept shutdown events and cancel themselves unless the creator of the timer specifies otherwise, and then for the timers that need to persist shutdown, specify that they should ignore shutdown.

For an example of how to change the timers, look at how execute process interacts with the shutdown event:

- https://github.com/ros2/launch/blob/34f10c69ea0f666348e11c2e5c7d66e947430730/launch/launch/actions/execute_process.py#L471-L474
- https://github.com/ros2/launch/blob/34f10c69ea0f666348e11c2e5c7d66e947430730/launch/launch/actions/execute_process.py#L240-L244

For an example of how the timer could cancel itself, look here:

- https://github.com/ros2/launch/blob/34f10c69ea0f666348e11c2e5c7d66e947430730/launch/launch/actions/execute_process.py#L343-L350

These are some of the timers that would need to be updated to ignore shutdown, but it's not an exhaustive list:

- https://github.com/ros2/launch/blob/34f10c69ea0f666348e11c2e5c7d66e947430730/launch/launch/actions/execute_process.py#L310-L331

pbaughman, 2019-02-09:
Thanks @wjwwood - I'll take a crack at this

pbaughman, 2019-02-15:
Merged!

closed 2019-02-15 by pbaughman