ros2/launch #186: [launch_ros] launch_ros.get_default_launch_description() exceptions

stonier, 2019-02-15:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - latest debs, 15/02/2019
- DDS implementation:
  - default
- Client library (if applicable):
  N/A

#### Steps to reproduce issue
Create a launch file:

```
#!/usr/bin/env python3
# -*- coding: utf-8 -*-

import launch_ros

def generate_launch_description():
    return launch_ros.get_default_launch_description()
```
#### Expected behavior

Launches without raising exceptions.

#### Actual behavior

```
[ERROR] [launch.LaunchService]: Caught exception in launch (see debug for traceback): Failed to init: rcl_init called on an already initialized context, at /tmp/binarydeb/ros-crystal-rcl-0.6.5/src/rcl/init.c:66
```

#### Additional information

Digging down into `get_default_launch_description()` and reproducing it without a `launch.LaunchDescription` entity, seems the problem is in `ROSSpecificLaunchStartup`.

wjwwood, 2019-02-15:
Sounds like we need to conditionally init in the launch code to avoid this error.

On the flip side, I'm not sure how to prevent users from running into issues if they're using launch for testing, but are doing init/shutdown in cycles from within their tests too. The shutdown will likely break launch in the process.

In C++, I recently added an option to have multiple simultaneous context's, which can each be init and shutdown separately. That would be a nice way to address this, as launch could create its own context so it does not interfere with, or is not interfered with by, others.

I don't remember off-hand if I properly refactored rclpy so that this is possible or not, I vaguely remember doing it, but let me check.

wjwwood, 2019-02-15:
Looks like you can:

https://github.com/ros2/rclpy/blob/174e3375d137f7239c9c83bebaf87c048e132bd5/rclpy/rclpy/__init__.py#L24-L28

So I think the right way to fix this would be to create a custom context in `ROSSpecificLaunchStartup` class and use that to avoid the reported exception and issues due to the user calling shutdown themselves.

stonier, 2019-02-18:
So I did some more digging and in the process of applying a custom context everywhere it turns out this is a red herring, at least for my case here.

In the example above, I was unwittingly making a second call to `launch_ros.get_default_launch_description()` in my custom `generate_launch_description()` method. 

The first call of course, is made in `ros2launch` itself ([here](https://github.com/ros2/launch/blob/300196b15b0aa99c045460e7d81333df99b8f8da/ros2launch/ros2launch/api/api.py#L141)). The examples in `launch_ros`, `ros2launch` are fine as well. Just bouncing between them, I'd mixed the convention for a standalone script with that of the ros2 launchable script.

----

For reference, passing a custom context to `rclpy.init()`, the `executor` and `create_node` seemed to do the trick. However, I don't have a gnarly use case yet to test it with such as that @wjwwood described. @wjwwood shall we close this until there is definitely a problem to be solved? The details will still be around for posterity.

wjwwood, 2019-02-19:
I'm fine with closing for now, I made a new issue to track the enhancement: https://github.com/ros2/launch/issues/189

closed 2019-02-19 by wjwwood