ros2/launch #187: setup_py / --symlink-install / launch files

stonier, 2019-02-18:
This came up in #83 and I'm interested in chasing it down. Only question is, where to start.

Example installation snippet for launch files in a pure setuptools python package:

```
    ....
    data_files=[
        ('share/' + package_name, ['package.xml', 'my.launch.py']),
    ],
    ....
```

Most of the heavy lifting seems to be done by [colcon-python-setup-py](https://github.com/colcon/colcon-python-setup-py) which does not seem to do anything out of the ordinary.

Curiously though, the `package.xml` gets symlinked but the launch files do not. In the installed dir:

```
launch/
package.bash
package.ps1
package.sh
my.launch.py
package.xml -> /my_workspace/src/my_package/package.xml
package.zsh
```

So, something else must be executing to symlink `package.xml`.

Any pointers on where to get started?

wjwwood, 2019-02-19:
Symlink install for python packages doesn't actually symlink anything, it uses setuptool's develop mode. So the files being installed with `data_files` are not actually being placed in the `install` folder. Also, launch has no way of finding the build or source folders for a package, so I don't have any ideas for how to improve that right now.

stonier, 2019-03-01:
Those files are actually being installed into the `install` folder, just not unwound and symlinked like `package.xml`. Talked to @dirk-thomas the other day though, he pointed to where colcon is doing the symlinking for python packages, so I'm shifting the details of this issue there (https://github.com/colcon/colcon-core/issues/169).

Wrapping up, workaround in the meantime (if you want symlink installs of launchers) is to just create standalone `ros2 run` style scripts for launchers and pull in the `launch_ros` functionality via library calls rather than use the `ros2 launch` framework.

closed 2019-03-01 by stonier