ros2/launch #189: [launch_ros] use a custom context to avoid interferring with user init/shutdown of rclpy

wjwwood, 2019-02-19:
## Feature request

#### Feature description

Right now `launch_ros` (or more specifically `ros2 launch`) will init rclpy, create a node, and then later shutdown rclpy. It also expects that shutdown is not called prematurely.

If the user did their own init or shutdown, it might interfere with this. So `launch_ros` should instead use a custom context and do its own init and shutdown there.

pbaughman, 2019-02-19:
@wjwwood I actually have a version of this done, but I'm waiting for ros2 crystal internally to test it (the version we have doesn't let you pass arguments to init yet).  I will push it for review, but I haven't been able to test it yet

wjwwood, 2019-02-20:
That would be great!

wjwwood, 2019-03-22:
This was done in #193.

closed 2019-03-22 by wjwwood