ros2/launch #194: add_test_action no longer triggers shutdown on completion

crdelsey, 2019-03-06:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - d5ace508426 (latest on 6 Mar 2019)
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Run `colcon test` on the navigation2 repo.

#### Expected behavior
Add the end of the test action execution, the `LaunchTestService` should terminate any other nodes that were launched.

#### Actual behavior
The nodes are left running until `ctest` times out the test execution.

#### Additional information

This change in behavior was introduced with PR #167. If I back that out, the expected behavior is restored.

Here's an example launch file that is timing out instead of terminating now. [dynamic_params_test.launch.py](https://github.com/ros-planning/navigation2/blob/master/nav2_dynamic_params/test/dynamic_params_test.launch.py)

There is some exception output. I'm not sure if it is related.

```
    Start 10: test_dynamic_params_client

10: Test command: /usr/bin/python3 "-u" "/home/crdelsey/src/ros_dev_workspace/ros2_ws/install/ament_cmake_test/share/ament_cmake_test/cmake/run_test.py" "/home/crdelsey/src/ros_dev_workspace/navigation2_ws/build/nav2_dynamic_params/test_results/nav2_dynamic_params/test_dynamic_params_client.xml" "--package-name" "nav2_dynamic_params" "--generate-result-on-success" "--env" "TEST_EXECUTABLE=/home/crdelsey/src/ros_dev_workspace/navigation2_ws/build/nav2_dynamic_params/test/test_dynamic_params_client_exec" "--command" "/home/crdelsey/src/ros_dev_workspace/navigation2_ws/navigation2/nav2_dynamic_params/test/dynamic_params_test.launch.py"
10: Test timeout computed to be: 60
10: -- run_test.py: extra environment variables:
10:  - TEST_EXECUTABLE=/home/crdelsey/src/ros_dev_workspace/navigation2_ws/build/nav2_dynamic_params/test/test_dynamic_params_client_exec
10: -- run_test.py: invoking following command in '/home/crdelsey/src/ros_dev_workspace/navigation2_ws/build/nav2_dynamic_params/test':
10:  - /home/crdelsey/src/ros_dev_workspace/navigation2_ws/navigation2/nav2_dynamic_params/test/dynamic_params_test.launch.py
10: [INFO] [launch]: process[test_dynamic_params_helper-1]: started with pid [27826]
10: [INFO] [launch]: process[test_dynamic_params_client-3]: started with pid [27827]
10: [INFO] [launch]: process[test_dynamic_params_client-3]: started with pid [27828]
10: [INFO] [launch]: process[test_dynamic_params_client-3]: process has finished cleanly
10: [INFO] [launch]: sending signal 'SIGINT' to process[test_dynamic_params_helper-1]
10: [INFO] [launch]: process[test_dynamic_params_client-3]: process has finished cleanly
10: [ERROR] [asyncio]: Task exception was never retrieved
10: future: <Task finished coro=<ExecuteProcess.__execute_process() done, defined at /home/crdelsey/src/ros_dev_workspace/ros2_ws/build/launch/launch/actions/execute_process.py:423> exception=InvalidStateError('invalid state',)>
10: Traceback (most recent call last):
10:   File "/home/crdelsey/src/ros_dev_workspace/ros2_ws/build/launch/launch/actions/execute_process.py", line 467, in __execute_process
10:     self.__cleanup()
10:   File "/home/crdelsey/src/ros_dev_workspace/ros2_ws/build/launch/launch/actions/execute_process.py", line 363, in __cleanup
10:     self.__completed_future.set_result(None)
10: asyncio.base_futures.InvalidStateError: invalid state
10: [INFO] [launch]: process[test_dynamic_params_helper-1]: process has finished cleanly
1/1 Test #10: test_dynamic_params_client .......***Timeout  60.01 sec
```

wjwwood, 2019-03-07:
@hidmic or @mjcarroll would one of you have some time to look at this?

hidmic, 2019-03-07:
Hmm for starters that example launch file is adding the same `test_dynamic_params_client` process action twice: once [here](https://github.com/ros-planning/navigation2/blob/master/nav2_dynamic_params/test/dynamic_params_test.launch.py#L45) and then [here](https://github.com/ros-planning/navigation2/blob/master/nav2_dynamic_params/test/dynamic_params_test.launch.py#L47). And I _think_ that's undefined behavior. We have a feature request, https://github.com/ros2/launch/issues/113, to prevent this from happening.

Anyways, @crdelsey may I ask you to provide the means to reproduce? A repos file with the relevant repositories would do.

crdelsey, 2019-03-07:
@hidmic Thanks for taking a look through that launch file. The duplicate addition of the of the test action was the problem. I appreciate the help :-)

I'll close the issue since it is resolved.

closed 2019-03-07 by crdelsey