ros2/launch #195: Support Python 3.5.2 (Xenial)?

jacobperron, 2019-03-06:
## Feature request

#### Feature description

Until recently, `launch` was compatible with Python 3.5.2, which is the current default version for `python3` on Ubuntu Xenial, but now is broken.

#167, introduced the following line:

https://github.com/ros2/launch/blob/d5ace5084262de96115ea7d68b1c43160b0bf3e7/launch/launch/actions/opaque_coroutine.py#L20

`Coroutine` is not available in the `typing` module until Python 3.5.3 ([reference](https://stackoverflow.com/questions/44651115/python-typing-module-missing-the-coroutine-class-in-python-3-5#comment76294521_44651115)).

I think it's an easy fix as we can use something like [TypeVar('Coroutine')](https://docs.python.org/release/3.5.2/library/typing.html#typing.TypeVar) as a replacement.
But, I'm not sure if we plan to support Xenial in Dashing and/or if we'd like to keep `launch` backwards compatible (given that it's an easy fix).

dirk-thomas, 2019-03-06:
Duplicate of ros2/build_cop#170.

jacobperron, 2019-03-07:
> Duplicate of ros2/build_cop#170.

@dirk-thomas Okay. What do you think about #196 as an alternate solution? I've been using it locally as I'm running Xenial. But I don't mind upgrading to Bionic if we're dropping support for Xenial.

jacobperron, 2019-03-07:
I've seen your comment in #196 :+1:

closed 2019-03-07 by jacobperron