ros2/launch #203: Introspector failing when env is set by the user

stonier, 2019-03-18:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - crystal
- DDS implementation:
  - default
- Client library (if applicable):
  - n/a

#### Steps to reproduce issue

Add an env dict to the arguments for a node action and also call the introspector.

```
    launch_description = launch.LaunchDescription()
    launch_description.add_action(
        launch_ros.actions.Node(
            package='py_trees_ros_tutorials',
            node_name="one",
            node_executable="tree-data-gathering",
            output='screen',
            env={'PYTHONUNBUFFERED': '1'}
        )
    )
    print(launch.LaunchIntrospector().format_launch_description(launch_description))
```

#### Expected behavior

No errors.

#### Actual behavior

```
Traceback (most recent call last):
  File "/mnt/mervin/workspaces/crystal/py_trees/install/py_trees_ros_tutorials/lib/py_trees_ros_tutorials/tutorial-one-data-gathering", line 11, in <module>
    load_entry_point('py-trees-ros-tutorials', 'console_scripts', 'tutorial-one-data-gathering')()
  File "/mnt/mervin/workspaces/crystal/py_trees/build/py_trees_ros_tutorials/py_trees_ros_tutorials/one_data_gathering.py", line 123, in launch_main
    print(launch.LaunchIntrospector().format_launch_description(launch_description))
  File "/opt/ros/crystal/lib/python3.6/site-packages/launch/launch_introspector.py", line 134, in format_launch_description
    entity_descriptions = format_entities(launch_description.entities)
  File "/opt/ros/crystal/lib/python3.6/site-packages/launch/launch_introspector.py", line 74, in format_entities
    result.extend(format_action(cast(Action, entity)))
  File "/opt/ros/crystal/lib/python3.6/site-packages/launch/launch_introspector.py", line 114, in format_action
    for k, v in typed_action.env.items()]) + '}',
AttributeError: 'list' object has no attribute 'items'
```

wjwwood, 2019-03-18:
Yeah, looks like the custom logic for ExecuteProcess hasn't been updated. My intention was to enable automatic rendering of launch entities, so that the action specific custom code in launch introspector could be removed at some point.

oswinso, 2019-03-26:
Weird, because I can't reproduce the error on Ubuntu 18.04 on Master. I get the correct output:
```
<launch.launch_description.LaunchDescription object at 0x7fdc7aa14240>
└── ExecuteProcess(cmd=[ExecInPkg(pkg='py_trees_ros_tutorials', exec='tree-data-gathering'), LocalVar('node name')], cwd=None, env={'PYTHONUNBUFFERED': '1'}, shell=False)```

stonier, 2019-03-27:
Yes, also confirmed. Looks like https://github.com/ros2/launch/commit/632ef1f3eca7dbc54fbed85ab65a5efc512d83ed#diff-970fc3c5d394ad33c14e4f9fe55288ceR113 bugfixed it, but this is applied post `v0.7.3` which is the version that this bug was reported against. I've tested that one-line patch on the crystal release and verified it works without anything beyond that line.

stonier, 2019-03-27:
@wjwwood this doesn't move in the direction you articulated, but it does resolve the bug. Further work could be outlined in a different issue?

nuclearsandwich, 2019-03-28:
CI for the branch in #220.

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

nuclearsandwich, 2019-03-28:
The failures above are down to https://github.com/ros2/launch/pull/169 not being backported. We've ignored them previously because launch wasn't being included in the patch release.

wjwwood, 2019-03-29:
So we need to backport #169 as well @nuclearsandwich?

wjwwood, 2019-03-30:
I backported #169 speculatively.

wjwwood, 2019-04-02:
I'm going to close this since it's fixed on master and it is staged for backport into Crystal as well.

commit 1b606dd2 (2019-03-30): [introspector] bugfix representation when env is set

Resolves #203.
commit 36f47cfe (2019-04-01): [introspector] bugfix representation when env is set (#222)

* [introspector] bugfix representation when env is set

Resolves #203.

* debugging DCO issues
closed 2019-04-02 by wjwwood