ros2/launch #205: [launch_ros] node_name overrides for a multi-node rclpy process

stonier, 2019-03-18:
#### Feature description

A means of setting `node_name` overrides for a multi-node rclpy process such as that given in #204.

#### Implementation considerations

This also brings into question the naming of the class `launch_ros.actions.Node` since it currently can launch a multi-node process. Perhaps that should fail on trying to launch a multi-node process and for that, use a `launch_ros.actions.MultiNode` action.

Note: This is a different problem to the composable node launch. Here the nodes are brought into the same process via ... the process,  not the launcher. I suspect the same situation would arise linking two c++ classes with nodes and instantiating them in the same application.

stonier, 2019-03-18:
FYI @IanTheEngineer

wjwwood, 2019-03-19:
https://answers.ros.org/question/316870/ros2-composition-and-node-names-with-launch-files/?answer=316925#post-id-316925