ros2/launch #206: TypeErorr: get_default_launch_description() got an unexpected keyword argument 'prefix_output_with_name'

huchhong, 2019-03-19:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu 18.04
- Installation type:
  from source
- Version or commit hash:
  commit: 849824be59e61480d39e302caa9fc815acce6b04
- DDS implementation:
   Fast-RTPS

#### Steps to reproduce issue
I downloaded ros2.repo from ros2 master branch, and the commit id of launch is given above. After i compiled ros2, i tested a simple launch.py which can be executed sucessfully in crystal release. Below is my launch.py:

```
import os
from launch import LaunchDescription
import launch.actions
import launch_ros.actions

def generate_launch_description():
    map_yaml_file = launch.substitutions.LaunchConfiguration('map', default="asdf")

    return LaunchDescription([
        launch_ros.actions.Node(
            package="param_test",
            node_executable="set_and_get_parameters",
            node_name="set_and_get_parameters",
            node_namespace="",
            output="screen",
            parameters=[{'map_file': map_yaml_file}]
        ),
        launch_ros.actions.Node(
            package="param_test",
            node_executable="access_other_node_parameters",
            node_name="access_other_node_parameters",
            node_namespace="",
            output="screen",
        )
    ])
```
Then i got this error: 

> TypeErorr: get_default_launch_description() got an unexpected keyword argument 'prefix_output_with_name'

#### Expected behavior

no error

sloretz, 2019-03-19:
Fixed by ros2/launch#201

hidmic, 2019-03-21:
To the best of my knowledge, this has been addressed by #201. @huchhong please confirm. I'll keep the issue closed for the time being.

closed 2019-03-21 by hidmic