ros2/launch #207: Launch hangs with mutiple ctrl-c

sumanth-nirmal, 2019-03-20:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04, 4.15.0-46-generic, python 3.5.2
- Installation type:
  - From source
- Version or commit hash:
  - Launch version `0.7.3`
- DDS implementation:
  - Fast-RTPS 
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

Multiple `ctrl-c` hangs ROS 2 launch system while exiting the launch file (with mutiple process).

```sh
$ ros2 launch lifecycle lifecycle_demo.launch.py
```

Then with mutiple `ctrl-c` hangs while exiting

```sh
[INFO] [launch]: process[lifecycle_talker-1]: started with pid [15428]
[INFO] [launch]: process[lifecycle_listener-2]: started with pid [15429]
[INFO] [launch]: process[lifecycle_service_client-3]: started with pid [15430]
[INFO] [lc_listener]: notify callback: Transition from state unconfigured to configuring
[INFO] [lc_talker]: on_configure() is called.
[INFO] [lc_listener]: notify callback: Transition from state configuring to inactive
[INFO] [lc_client]: Transition 1 successfully triggered.
[INFO] [lc_client]: Node lc_talker has current state inactive.
[INFO] [lc_talker]: Lifecycle publisher is currently inactive. Messages are not published.
[WARN] [LifecyclePublisher]: Trying to publish message on the topic '/lifecycle_chatter', but the publisher is not activated
^C[WARNING] [launch.LaunchService]: user interrupted with ctrl-c (SIGINT)
^C[WARNING] [launch.LaunchService]: user interrupted with ctrl-c (SIGINT) again, ignoring...
[INFO] [launch]: process[lifecycle_talker-1]: process has finished cleanly
[INFO] [launch]: process[lifecycle_listener-2]: process has finished cleanly
[ERROR] [launch]: process[lifecycle_service_client-3] process has died [pid 15430, exit code -2, cmd '/opt/ApexOS/lib/lifecycle/lifecycle_service_client'].
^C[WARNING] [launch.LaunchService]: user interrupted with ctrl-c (SIGINT) again, ignoring...
^C[WARNING] [launch.LaunchService]: user interrupted with ctrl-c (SIGINT) again, ignoring...
^C[WARNING] [launch.LaunchService]: user interrupted with ctrl-c (SIGINT) again, ignoring...
^C[WARNING] [launch.LaunchService]: user interrupted with ctrl-c (SIGINT) again, ignoring...
^C[WARNING] [launch.LaunchService]: user interrupted with ctrl-c (SIGINT) again, ignoring...
^C[WARNING] [launch.LaunchService]: user interrupted with ctrl-c (SIGINT) again, ignoring...
^C[WARNING] [launch.LaunchService]: user interrupted with ctrl-c (SIGINT) again, ignoring...
^C[WARNING] [launch.LaunchService]: user interrupted with ctrl-c (SIGINT) again, ignoring...

^C[WARNING] [launch.LaunchService]: user interrupted with ctrl-c (SIGINT) again, ignoring...
^C[WARNING] [launch.LaunchService]: user interrupted with ctrl-c (SIGINT) again, ignoring...
^C[WARNING] [launch.LaunchService]: user interrupted with ctrl-c (SIGINT) again, ignoring...
^C[WARNING] [launch.LaunchService]: user interrupted with ctrl-c (SIGINT) again, ignoring...

^C[WARNING] [launch.LaunchService]: user interrupted with ctrl-c (SIGINT) again, ignoring...

```

#### Expected behavior

The Launch should gracefully exit even with mutiple `ctrl-c`

#### Actual behavior

The Launch hangs while exiting with mutiple `ctrl-c`

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

koonpeng, 2019-07-09:
I have the same problem, additionally, it sometimes result in defunct/zombie processes, looks like a race condition somewhere. The snippet code tries to recreate the zombie processes scenario.

```
import os
import signal
import subprocess
import time

proc = subprocess.Popen(['ros2', 'launch', 'lifecycle', 'lifecycle_demo.launch.py'], start_new_session=True)
pgid = os.getpgid(proc.pid)
time.sleep(1)
os.killpg(pgid, signal.SIGINT) # mimic bash ctrl+c behaviour
time.sleep(0.001)
os.killpg(pgid, signal.SIGINT) # mimic bash ctrl+c behaviour
proc.wait()
```

ivanpauno, 2019-07-09:
@sumanth-nirmal Ubuntu 16.04 is not a supported platform since Dashing (see http://www.ros.org/reps/rep-2000.html#dashing-diademata-may-2019-may-2021). Probably the problem is specific to that platform.

I couldn't reproduce the problem (Ubuntu 18.04, python 3.6.7, from source).

A simple `cntrl-c` works fine to me (return code 0), double `cntrl-c` shows a long traceback (return code 130):
```
Task exception was never retrieved                                                                                             [3/67]
future: <Task finished coro=<LaunchService.__run_loop() done, defined at /home/ivanpauno/ros2_ws/build/launch/launch/launch_service.p
y:206> exception=KeyboardInterrupt()>
Traceback (most recent call last):
  File "/home/ivanpauno/ros2_ws/build/launch/launch/launch_service.py", line 350, in run
    self.__loop_from_run_thread.run_until_complete(run_loop_task)
  File "/usr/lib/python3.6/asyncio/base_events.py", line 460, in run_until_complete
    self.run_forever()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 427, in run_forever
    self._run_once()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 1440, in _run_once
    handle._run()
  File "/usr/lib/python3.6/asyncio/events.py", line 145, in _run
    self._callback(*self._args)
  File "/home/ivanpauno/ros2_ws/build/launch/launch/launch_service.py", line 240, in __run_loop
    await process_one_event_task
  File "/home/ivanpauno/ros2_ws/build/launch/launch/launch_service.py", line 350, in run
    self.__loop_from_run_thread.run_until_complete(run_loop_task)
  File "/usr/lib/python3.6/asyncio/base_events.py", line 460, in run_until_complete
    self.run_forever()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 427, in run_forever
    self._run_once()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 1440, in _run_once
    handle._run()
  File "/usr/lib/python3.6/asyncio/events.py", line 145, in _run
    self._callback(*self._args)
  File "/home/ivanpauno/ros2_ws/build/launch/launch/launch_service.py", line 177, in _process_one_event
    await self.__process_event(next_event)
  File "/home/ivanpauno/ros2_ws/build/launch/launch/launch_service.py", line 186, in __process_event
    entities = event_handler.handle(event, self.__context)
  File "/home/ivanpauno/ros2_ws/build/launch/launch/event_handlers/on_shutdown.py", line 67, in handle
    return self.__on_shutdown(cast(Shutdown, event), context)
  File "/home/ivanpauno/ros2_ws/build/launch_ros/launch_ros/default_launch_description.py", line 38, in _shutdown
    self.__rclpy_spin_thread.join()
  File "/usr/lib/python3.6/threading.py", line 1056, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.6/threading.py", line 1072, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
  File "/home/ivanpauno/ros2_ws/build/launch/launch/utilities/signal_management.py", line 119, in __on_sigint
    __original_sigint_handler(signum, frame)
KeyboardInterrupt
```

@koonpeng Could you share your system details?

koonpeng, 2019-07-10:
CPU: Ryzen 2700x
Operating System: ubuntu 18.04, 4.15.0-54-generic
ros-version: Crystal
Installation from: apt
ros2 launch: 0.7.4-0bionic.20190402.213022
python: 3.6.8

I get the KeyboardInterrupt error sometimes too, depending on the timing between the 2 SIGINTs.

I should add that I ran into this problem when running ros2 launch from a script. The script itself sends SIGINT to ros2 launch, but bash also sends SIGINT to it as well when you press ctrl+c. This results in the double ctrl+c bug.

ivanpauno, 2019-07-10:
> CPU: Ryzen 2700x
> Operating System: ubuntu 18.04, 4.15.0-54-generic
> ros-version: Crystal
> Installation from: apt
> ros2 launch: 0.7.4-0bionic.20190402.213022
> python: 3.6.8
>
> I get the KeyboardInterrupt [...]

I tried the script and I never got a zombie process, or ctrl-c ignored and the process hanging.
It seems that we have something fishy with signal handling, though.

nuclearsandwich, 2019-12-05:
@sumanth-nirmal do you have any information to help us investigate this further?

In the meantime I am going to remove this from the Dashing project boards until a reliable reproduction can be found.

sumanth-nirmal, 2019-12-19:
@nuclearsandwich I tried with Dashing on Ubuntu 18.04 and can't reproduce this problem. I think we can close the issue.

nuclearsandwich, 2019-12-19:
Great. Thanks for checking.

closed 2019-12-19 by nuclearsandwich

Aposhian, 2024-08-13:
Reproduced this on Ubuntu 22.04 on Humble. It seems rare, but I think it just comes from multiple Ctrl+C's in a row.