ros2/launch #214: consider changing default output to "both"

dirk-thomas, 2019-03-21:
The current default is [log](https://github.com/ros2/launch/blob/86f663db2ff878b5541bf4a51a23f04eac606f5f/launch/launch/actions/execute_process.py#L92) and defined on a per process level. It would be more convenient if the default would be `both` since the common user expects to see the output of the processes.

Independent of the default value there should be a command line option to change the global default.

wjwwood, 2019-03-21:
> 

That's because of how it is in ROS 1:

> output="log|screen"(optional)
>    If 'screen', stdout/stderr from the node will be sent to the screen. If 'log', the stdout/stderr output will be sent to a log file in $ROS_HOME/log, and stderr will continue to be sent to screen. The default is 'log'. 

-- from: https://wiki.ros.org/roslaunch/XML/node

I don't know if `both` is a better default, but my personal preference is to see everything. That does go in contrast other defaults, like rosbuild and colcon where the default is to hide the build output and only show errors.

> Independent of the default value there should be a command line option to change the global default.

I don't know what the appropriate way to do this is.

I think the best way for a user to do this would be to have a `launch.actions.DeclareLaunchArgument` which added an argument to do this, and then have `launch.actions.ExecuteProcess` check the corresponding `launch.substitutions.LaunchConfiguration` and let it override the user provided setting for `output`.

So perhaps we could have that argument action included by default, perhaps in this:

https://github.com/ros2/launch/blob/86f663db2ff878b5541bf4a51a23f04eac606f5f/launch_ros/launch_ros/default_launch_description.py#L88-L91

dirk-thomas, 2019-03-21:
> I don't know if both is a better default, but my personal preference is to see everything. That does go in contrast other defaults, like rosbuild and colcon where the default is to hide the build output and only show errors.

Good point with the comparison of other tools. I understand the desire to keep the verbosity low. With #212 being fixed the urgency for default verbosity is much less important to me.

> I think the best way for a user to do this would be ...

Sounds good to me. Would it be possible to see this option when calling the launch file with `--help` then?

dirk-thomas, 2019-03-21:
I changed the title to reflect the direction of the conversation and unassigned the ticket for now leaving it as an enhancement.

wjwwood, 2019-03-21:
> Sounds good to me. Would it be possible to see this option when calling the launch file with --help then?

There's a `--show-arguments (-s)` which will show you the launch arguments. But we could add an option to `ros2launch` which automatically passes this argument to the launch file, basically just using the launch argument as the mechanism to interact with the launch file.

The other way to do it would be to use an environment variable.

hidmic, 2019-03-25:
Hmm, shouldn't the argument be available regardless of whether one is using ROS specific launch entities or not? Perhaps we need a `default_launch_description` that is ROS agnostic?

wjwwood, 2019-03-25:
That would be nice to have, but you'd need to update `ros2launch` to take that option as well.

Honestly, an environment variable might be just as good or better.

hidmic, 2019-03-25:
I have no preference really, so it's up to you guys how/if you want to set a global default for launched processes output configuration. @dirk-thomas ?

dirk-thomas, 2019-03-25:
I don't have a preference either- I would just like to be able to change the configuration somehow.

wjwwood, 2019-03-25:
The environment variable is less intrusive, but also not automatically documented in the launch description (not as visible as a declare launch argument action). However, the main purpose is to use it with `ros2launch` and that could have a command line argument which is documented and which just sets the environment variable locally if asked to by the command line flag. So I'm leaning to the env var.

To recap, I'm suggesting we:

- add a command line option to `ros2 launch` like `--show-all-subprocess-output`
  - (or something similar, name could be improved...)
- if set, that command line argument would set an environment variable of a similar name
- ExecuteProcess would look at that environment variable during construction, allowing it to override the output destination to include the screen

What do you think @hidmic?

hidmic, 2019-03-25:
Sounds good to me. How do you feel about documenting the equivalent envvar along with the new `ros2 launch` command line option?

wjwwood, 2019-03-25:
I think that's a good idea, it should be documented in the `ExecuteProcess` class as well, but explaining that the command line argument works by using the env var is a good idea.

PR #218 (2019-03-26): Support external ExecuteProcess output overrides.
Fixes #214 by means of an `$OVERRIDE_LAUNCH_PROCESS_OUTPUT` envvar. 

Lacking a test to keep this PR simple, we can add one if necessary.
closed 2019-03-26 by hidmic