ros2/launch #248: [launch] Unexpected failure when providing arguments to an included launch description

mxgrey, 2019-05-21:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Binaries
- Version or commit hash:
  - Crystal
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Create a simple python ROS2 package (I'm naming it `launch_test`) containing these two launch files:

```
# launch/parent.launch.py

import launch

def generate_launch_description():
    return launch.LaunchDescription([
        launch.actions.DeclareLaunchArgument('my_arg')
    ])
```

```
# launch/child.launch.py

import launch
from launch.actions.include_launch_description import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions.this_launch_file_dir import ThisLaunchFileDir

from ament_index_python import get_package_share_directory

def generate_launch_description():

    return launch.LaunchDescription([
        IncludeLaunchDescription(
            PythonLaunchDescriptionSource([get_package_share_directory('launch_test'), '/parent.launch.py']),
            launch_arguments={'my_arg': 'my_value'}.items()
        )
    ])
```

Then build/install this package to a workspace and run

```
$ ros2 launch -d launch_test child.launch.py
```

#### Expected behavior

The `parent.launch.py` launch description should recognize that it's been provided with the `my_arg` argument when it was included by `child.launch.py`.

#### Actual behavior

An exception occurs:

```
RuntimeError: Included launch description missing required argument 'my_arg' (description: 'no description given'), given: []
```

It seems that the launch service doesn't recognize that `parent.launch.py` has been provided with the launch argument by `child.launch.py`.

#### Additional information

It's possible to work around this by giving a default value for `my_arg` in `parent.launch.py`, but that's undesirable if I want to enforce that the argument has provided externally (whether it's provided from the command line or by a child launch file should be irrelevant, ideally).

mxgrey, 2019-05-21:
I've just realized a better workaround for anyone that runs into this issue. Just explicitly declare the launch argument in the child launch file description:

```
def generate_launch_description():

    return launch.LaunchDescription([
        launch.actions.DeclareLaunchArgument('my_arg', default_value='my_value'),
        IncludeLaunchDescription(
            PythonLaunchDescriptionSource([get_package_share_directory('launch_test'), '/parent.launch.py']),
            launch_arguments={'my_arg': 'my_value'}.items()
        )
    ])
```

But this strikes me as not being ideal, since (1) it's duplicating information, and (2) it implies that `my_arg` is meant to be set by the user, which is not really the case.

ivanpauno, 2019-05-21:
I have been able to reproduced this.

Additional info: Strangely, if you create the LaunchService in the child.launch.py, include the description and run it, the problem does not appear.

I will try to dig a little bit more into this.

PR #249 (2019-05-28): Do not check declared arguments of IncludeLaunchDescription subentities
Closes #248
closed 2019-05-28 by ivanpauno