ros2/launch #253: remove conceptual dependency on ROS_DOMAIN_ID in launch_testing

wjwwood, 2019-05-29:
Currently `launch_testing` has options/features related to `ROS_DOMAIN_ID`:

https://github.com/ros2/launch/blob/066009b097e06dad7f981de55b9ee05c83cf6e0b/launch_testing/launch_testing/launch_test.py#L51

This should not be the case, it is inappropriate for anything in this repository to depend on, reference, or really even mention anything from ROS.

There's a TODO, which I linked to, but I opened this issue to raise the severity. This needs to be changed.

@hidmic @pbaughman please don't let things like this leak into `launch` or `launch_testing` in the future.

pbaughman, 2019-05-30:
Do you have any suggestions about how to fix it?  The cure (removing this) seems worse than the disease right now

pbaughman, 2019-05-30:
Could we add an extension point to launch_test so that when launch_testing_ros is installed you get the isolation behavior?

pbaughman, 2019-05-30:
I guess we could also add a different runner with a different name to run tests that need ROS

hidmic, 2019-05-30:
> This should not be the case, it is inappropriate for anything in this repository to depend on, reference, or really even mention anything from ROS.

I fully agree. As it stands, this option could be moved to the [`ros2test`](https://github.com/ros2/ros_testing/blob/master/ros2test/ros2test/command/test.py#L24) CLI command with little to no effort (owing to the fact it's just an envvar).

pbaughman, 2019-05-30:
@hidmic will we add a different cmake function that calls the ros2cli command?  The dependency graph for that cmake function will be quite long, but maybe not that much longer than launch_ros

hidmic, 2019-05-30:
@pbaughman there is one already, [`add_ros_test()`](https://github.com/ros2/ros_testing/blob/master/ros_testing/cmake/add_ros_test.cmake).

pbaughman, 2019-05-30:
@hidmic I want to refactor the domain coordinator into its own package so I can do this https://github.com/ament/ament_cmake/issues/172 so lets's make sure we coordinate on this.

ivanpauno, 2019-06-04:
> @hidmic I want to refactor the domain coordinator into its own package so I can do this [ament/ament_cmake#172](https://github.com/ament/ament_cmake/issues/172) so lets's make sure we coordinate on this.

Sounds good. Where should the domain coordinator live?
If it doesn't fit well in an existing repo, I will need a new one (I can't create one).

pbaughman, 2019-06-04:
@ivanpauno In the WIP I've got now, it lives in ament_cmake_ros along with a new ament_cmake_pytest_isolated and an ament_cmake_gtest_isolated package

ivanpauno, 2019-06-04:
> @ivanpauno In the WIP I've got now, it lives in ament_cmake_ros along with a new ament_cmake_pytest_isolated and an ament_cmake_gtest_isolated package

I think `ament_cmake_pytest_isolated` and `ament_cmake_gtest_isolated` should live in `ament_cmake_ros`. But the code of the domain coordinator could be in a separated repo, as it is really a python package (and not cmake stuff). It's ok for me having it there though.

pbaughman, 2019-06-04:
@ivanpauno Let's talk about it in the MR once I open it, but there's one cmake thing that I jammed into the domain_coordinator package - it might make sense to move it, or it might not.

PR #256 (2019-06-10): Remove launch_testing dependency on ros domain ID
Fixes #253.

Depends on https://github.com/ros2/ros_testing/pull/2.
closed 2019-06-10 by ivanpauno