ros2/launch #287: Support for respawn in `ExecuteProcess`

ivanpauno, 2019-07-25:
## Feature request

Add support for `respawn` and `respawn_delay`.

#### Feature description

ROS 1 supported those attributes in `node` tag. We could have something similar in `ExecuteProcess` action. The behavior could be emulated with event handlers, but having an argument in the constructor for that may be easier to use.

There are some point of discussions:
- We could just implement it for the frontend implementations (XML, YAML), and avoid adding an argument to `ExecuteProcess` constructor.
- Should event handlers targeting the original process target the new one?

Previous discussion: https://github.com/ros2/ros2_documentation/pull/302#discussion_r306309349

ivanpauno, 2019-07-25:
@wjwwood @hidmic FYI.

> We could just implement it for the frontend implementations (XML, YAML), and avoid adding an argument to ExecuteProcess constructor.

I prefer implementing it in `ExecuteProcess`.

> Should event handlers targeting the original process target the new one?

I think they should. I don't see an use case were that isn't desired.

mkhansen-intel, 2019-10-10:
+1 for getting this feature, is it planned for anytime soon?

wjwwood, 2019-10-10:
> +1 for getting this feature, is it planned for anytime soon?

I think the best assumption is no. Maybe @hidmic or @ivanpauno were planning on doing it, but I'm full booked up until well after ROSCon.

wjwwood, 2019-10-10:
> It would also be nice if we could signal somehow that the node has been respawned to the lifecycle_manager node so that it could re-activate any node that crashes. Not to digress too far, but is there a mechanism for doing something like that?

Assuming it knows the node's name, it could look for the node to enter the unconfigured state (or whatever the first lifecycle event published is) on the event topic for that node. Launch does the same thing in order to produce launch events for each state transition.

ivanpauno, 2019-10-10:
> +1 for getting this feature, is it planned for anytime soon?

I'm not planning to work on this in a near future.
If someone wants to work on it, I can help doing reviews.

mkhansen-intel, 2019-10-10:
> Assuming it knows the node's name, it could look for the node to enter the unconfigured state (or whatever the first lifecycle event published is) on the event topic for that node. Launch does the same thing in order to produce launch events for each state transition.

That makes sense. We'll discuss that as a potential way of doing this. It won't matter until the 'respawn' capability is available though.

Jconn, 2020-02-14:
has anything changed here? This is a nice to have for us

iuhilnehc-ynos, 2020-06-03:
@ivanpauno @hidmic 
I'd like to contribute to this feature.

ivanpauno, 2020-06-03:
Thanks @iuhilnehc-ynos !
Let us know if you have any question or when you need a review.

fujitatomoya, 2020-06-04:
@iuhilnehc-ynos 

can you make PR against this? so that they can start review.

iuhilnehc-ynos, 2020-06-05:
@ivanpauno @fujitatomoya 

I have made a PR for this feature, please review https://github.com/ros2/launch/pull/426 .

fujitatomoya, 2020-06-23:
@ivanpauno 

friendly ping, i think that you could close this one.

ivanpauno, 2020-06-24:
Fixed in #426, thanks for the contribution @iuhilnehc-ynos !

closed 2020-06-24 by ivanpauno

tuxnese, 2021-01-21:
Is there a plan to port to Foxy?

fujitatomoya, 2021-01-22:
i think currently no. this is feature, not bug fix. so it is unlikely to happen, i think.

tuxnese, 2021-01-22:
I have found the PR from blacksoul000:foxy . I have tried the blacksoul000:foxy fork and it seems to work fine, although I have not yet tested it a lot.

tonynajjar, 2021-11-15:
Hi @mkhansenbot, @wjwwood 

> That makes sense. We'll discuss that as a potential way of doing this. It won't matter until the 'respawn' capability is available though.

Now that respawn is available, was respawning + activating ever discussed again? Would be useful for me. Workarounds are also appreciated