ros2/launch #303: Conditional includes

chapulina, 2019-08-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Bionic
- Installation type:
  - debs

#### Steps to reproduce issue

While working on https://github.com/ros-simulation/gazebo_ros_pkgs/pull/962, we wanted to conditionally add the `IncludeLaunchDescription` action using the `condition` keyword. But that action is not accepting the keyword.

Here's a simple example with 2 files:

*main.launch.py*
```
from launch import Condition
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.actions import ExecuteProcess
from launch.actions import IncludeLaunchDescription
from launch.conditions import IfCondition
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import LaunchConfiguration
from launch.substitutions import ThisLaunchFileDir

def generate_launch_description():
    return LaunchDescription([
        DeclareLaunchArgument('do', default_value='true',
                              description='false == dont'),
        ExecuteProcess(
            cmd=['echo',  'main'], output='screen',
            condition=IfCondition(LaunchConfiguration('do'))),

        IncludeLaunchDescription(
            PythonLaunchDescriptionSource([ThisLaunchFileDir(), '/included.launch.py']),
            condition=IfCondition(LaunchConfiguration('do'))
        ),
    ])

```
*included.launch.py*
```
from launch import LaunchDescription
from launch.actions import ExecuteProcess

def generate_launch_description():
    return LaunchDescription([
        ExecuteProcess(
            cmd=['echo',  'included'], output='screen')
    ])
```

#### Expected behavior

`main.launch.py` prints

```
[echo-1] main
[echo-2] included
```

`main.launch.py do:=false` prints nothing

#### Actual behavior

`[ERROR] [launch]: Caught exception in launch (see debug for traceback): __init__() got an unexpected keyword argument 'condition'`

#### Additional information

Commenting out the `condition` inside `IncludeLaunchDescription` lets the program run, but `included` is always printed.

hidmic, 2019-08-14:
Yeap, that's a bug. [Current implementation](https://github.com/ros2/launch/blob/master/launch/launch/actions/include_launch_description.py#L73) is not forwarding (nor taking) additional `kwargs` to its base class constructor.

chapulina, 2019-08-14:
I haven't checked, but this may be also true for other actions. It may be a good idea to go over the arguments for all actions.

sloretz, 2019-08-14:
> Yeap, that's a bug. Current implementation is not forwarding (nor taking) additional kwargs to its base class constructor.

@chapulina and I tried making a subclass of `IncludeLaunchDescription` that forwarded args to the `Action` constructor, but more needs to be done. `IncludeLaunchDescription` [overrides `visit(...)`](https://github.com/ros2/launch/blob/884e14e689a2a1cd75120dc6803d320c327f9e1f/launch/launch/actions/include_launch_description.py#L125), which is what [evaluates the condition](https://github.com/ros2/launch/blob/884e14e689a2a1cd75120dc6803d320c327f9e1f/launch/launch/action.py#L88) in the base class.

hidmic, 2019-08-14:
True that, though I don't see why it cannot override `execute(...)` instead. I wonder if this class was written before `Action` had any features. In any case, yes, it needs to be fixed at the source (as opposed to in a subclass).

closed 2019-08-15 by hidmic