ros2/launch #305: Change how launch_testing waits for the launched system to be ready

pbaughman, 2019-08-14:
## Feature request
Change the ready_fn in generate_test_description into a launch action

#### Feature description

The original intent of the ready_fn was to avoid an issue frequently encountered with ros1's rostest where each test would have to contain logic to wait for the launched system to be ready to be interacted with.  The purpose of the ready_fn is to force the test author to clearly indicate when it's safe for the tests to start to run.

What I'd like to do is change the ready_fn from a function passed into generate_test_description to something that's included in the launch description itself.  This would allow launch_testing to introspect the launch description and make sure that the action was included, as well as give a better error message if the element is omitted.  

Starting with the [good_proc_test example](https://github.com/ros2/launch/blob/master/launch_testing/examples/good_proc.test.py) I'm proposing that after this change, it would be possible to do this instead:

```
def generate_test_description():
    return launch.LaunchDescription([
        dut_process,

        # Start tests right away - no need to wait for anything
        launch_testing.actions.ReadyToTest(),  # <--- I'm not married to this name
    ])
```

This should simplify the launch_testing API a bit, and maybe make it easier to write a pytest plugin in the future instead of needing a special runner.

#### Implementation considerations

I can keep support for the old way pretty easily, so we won't break anything.

I would have the test_runner call generate_test_description to get the launch description and the test context, just like it does now.  Then introspect the launch description for a ReadyToTest action and plumb that to the python event it's currently using to delay the start of the tests

pbaughman, 2019-08-14:
@hidmic Any idea who else would have input on this?  I'll wait a little before starting, but I think this is maybe only a few hours of work for me.  I think it's incremental progress towards a simpler API, but maybe someone else has a grand plan that's different.

wjwwood, 2019-08-30:
This seems reasonable to me. I don't know if it will make integrating with pytest easier or not, but it seems like it wouldn't make it harder.

PR #322 (2019-09-20): Optionally remove ready fn arg from generate_test_description
Resolves #305

This change lets test launch descriptions include a launch_testing.actions.ReadyToTest action to signal when it's safe for tests to start.

This is intended to be used instead of passing a ready_fn into the generate_test_description function which then has to be plumbed into the launch description with an launch.actions.OpaqueFunction

Longer term, I would like to deprecate the use of the ready_fn in favor of the ReadyToTest action.  The former is annoying to plumb especially in tests where the test launch description [... cut]
closed 2019-09-20 by wjwwood