ros2/launch #311: Nodes launch API

J-pcy, 2019-08-16:
I tried to understand how to launch multiple nodes using python or yaml file, but I did not find anything related to rclpy/rclcpp to create a node and pass parameters in launch repository. May I ask how it works? Thank you.

ivanpauno, 2019-08-16:
It's in [launch_ros repo](https://github.com/ros2/launch_ros):
https://github.com/ros2/launch_ros/blob/master/launch_ros/launch_ros/actions/node.py.

Consider doing general questions in [ROS Answers](https://answers.ros.org).

ivanpauno, 2019-08-16:
This tutorials can also help:
- https://index.ros.org/doc/ros2/Tutorials/Launch-system/
- https://index.ros.org/doc/ros2/Tutorials/Composition/#composition-using-launch-actions

closed 2019-08-16 by ivanpauno

J-pcy, 2019-08-16:
Sorry for that and thanks for your answer.