ros2/launch #325: [regression] 'GroupAction' object has no attribute 'actions'

orduno, 2019-08-30:
## Bug report

**Required Info:**

- Operating System: Ubuntu 18.04
- Installation type: source
- Version or commit hash: cb911b9af331c0b17e2c3b0dec375fa95a190548
- DDS implementation: Fast-RTPS

#### Steps to reproduce issue

Launch the following script:

```python

from launch import LaunchDescription
from launch.actions import GroupAction
from launch.actions import LogInfo

import launch.actions
import launch_ros.actions

def generate_launch_description():
    return LaunchDescription([
        GroupAction([
            launch.actions.LogInfo(msg=['Doing Action 1'])
        ])
    ])

```

#### Expected behavior
Terminal output with message `Doing Action 1`.

#### Actual behavior
Terminal output is:

``` bash
[ERROR] [launch]: Caught exception in launch (see debug for traceback): 'GroupAction' object has no attribute 'actions'
```

orduno, 2019-08-30:
This was introduced after 0cf52d1ac9ce2a51205e13b8220c255bde38618f

orduno, 2019-09-03:
@ivanpauno I think the issue was introduced in #310.

where:

https://github.com/ros2/launch/blob/e0a13cea31f9439e96e9baa6b54d79c5acd9dad5/launch/launch/actions/group_action.py#L74

should be:

```python
return self.__actions
```

ivanpauno, 2019-09-03:
> @ivanpauno I think the issue was introduced in #310.
> 
> where:
> 
> https://github.com/ros2/launch/blob/e0a13cea31f9439e96e9baa6b54d79c5acd9dad5/launch/launch/actions/group_action.py#L74
> 
> should be:
> 
> ```python
> return self.__actions
> ```

My bad. I was sure that we had a test for `GroupAction`, it seems that we don't.
I will work correct the error and add a test.

ivanpauno, 2019-09-03:
> I was sure that we had a test for GroupAction, it seems that we don't.

There is actually a test, but it isn't complete enough.

PR #327 (2019-09-04): Fix "'GroupAction' object has no attribute 'actions'" error
Fixes #325
closed 2019-09-04 by ivanpauno