ros2/launch #337: [frontend] FindPackage substitution not concatenated with surrounding text

jacobperron, 2019-09-23:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - 897cae8f6bf5602e250f9e521497b9eb70e14a85
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Create a launch file with the following content and launch it.
```
<launch>
  <executable cmd="echo $(find-pkg-share rclcpp)/sub/path/foo" output="screen"/>
</launch>
```

#### Expected behavior

I would expect to see output like the following:

    [echo-2] /home/jacob/ros2_ws/install/rclcpp/share/rclcpp/sub/path/foo

#### Actual behavior

The substituted string for the share directory and the subsequent text, `/sub/path/foo`, are separated with a space:

    [echo-2] /home/jacob/ros2_ws/install/rclcpp/share/rclcpp /sub/path/foo

#### Additional information

There's the same behavior with `find-pkg-prefix` as well.

I've noticed that if we do the substitution in an `<arg>` tag, then it works as expected.
I'm not sure if tags other than `<executable>` are exhibiting the same bug.

#### Workaround

First do the substitution in an `<arg>` tag, for example:

```
<launch>
  <arg name="rclcpp_sub_dir" default="$(find-pkg-share rclcpp)/sub/path/foo" />
  <executable cmd="echo $(var rclcpp_sub_dir)" output="screen"/>
</launch>
```

---

I'm not sure where to look to resolve this issue. It's possible it should be fixed in `launch` (not `launch_ros`).
Maybe someone can point me in the right direction.

ruffsl, 2020-02-05:
Has this moved since sep? Is there branch with some WIP related to this issue? This is still effecting launch users, limiting the use of substitutions.

ivanpauno, 2020-02-06:
@ruffsl Yes, I agree. I will try to iterate on this ASAP.

PR #379 (2020-02-17): Fix parsing of cmd line arguments in XML and yaml file
Fixes https://github.com/ros2/launch/issues/337.

The rules are a bit tricky, but I added a bunch of test cases. I will add any other suggested test case if needed.

I first though about solving the problem by addressing https://github.com/ros2/launch/issues/263.
But I later realized that:
- That change breaks API and it's not backportable. This fixes #337 in a backportable way.
- Using a string (`SomeSubstitutionsType`) for specifying commands, instead of a list (`List[SomeSubstitutionsType]`), makes concatenation of arguments more comple [... cut]
closed 2020-02-17 by ivanpauno