ros2/launch #338: [launch_frontend] Produce errors for invalid entities and attributes

jacobperron, 2019-09-23:
## Feature request

#### Feature description
When an invalid entity or attribute appears in a launch file, it should produce an error.
This way users are notified of mistakes in their launch files instead of them being silently ignored.
For instance, it is easy to mistype the attribute `output`.

Here's an example launch file that I would expect to fail for a couple of reasons:

```xml
<launch>
  <!-- The attribute 'outputs' should cause an error -->
  <node
    pkg="demo_nodes_cpp"
    exec="talker"
    outputs="screen"
    name="talker">
    <!-- This tag should cause an error -->
    <this_is_not_a_valid_entity />
  </node>
</launch>
```

#### Implementation considerations

Launch files already throw errors for invalid entities at the global scope. E.g. the following launch file will error:

<launch>
  <this_is_not_a_valid_entity />
</launch>

Maybe we can reuse the logic for nested scopes as well.

ivanpauno, 2020-10-28:
This issue is somewhat related to https://github.com/ros2/launch/issues/392, though not exactly the same.

ivanpauno, 2020-10-29:
I have some ideas on how to achieve this, I'm assigning self to try them.

closed 2020-11-04 by ivanpauno