ros2/launch #340: Why is python_cmake_module needed by launch_testing_ament_cmake, and can we skip it?

pbaughman, 2019-09-23:
Preface: I think this is a question for @hidmic but I'm open to input from anybody.

#### Background
I see that launch_testing_ament_cmake [requires python_cmake_module](https://github.com/ros2/launch/blob/master/launch_testing_ament_cmake/launch_testing_ament_cmake-extras.cmake#L17)  I traced this back to when we merged apex_launchtest into launch_testing, but I think this was not part of apex_launchtest when it was merged.

My understanding is that this package is required to make [this line work](https://github.com/ros2/launch/blob/master/launch_testing_ament_cmake/cmake/add_launch_test.cmake#L66) on Windows.

#### My problem
We have an environment where this package isn't available at build time, and it's preventing us from building packages that use 'launch_testing_ament_cmake' which is now quite a few packages in 'Dashing'

#### My Question
Since the use of PYTHON_EXECUTABLE_DEBUG is `WIN32` only, would it be possible to enclose the `find_package` in an `IF(WIN32)` block, or does this only solve half the problem?  I see that this package is also a `<buildtool_export_depend>python_cmake_module</buildtool_export_depend>` in the package.xml

Can someone shed some light on why this is used in WIN32?  Just trying to understand the history of this being added.

I notice that ament_cmake_pytest [doesn't have this dependency](https://github.com/ament/ament_cmake/blob/master/ament_cmake_pytest/ament_cmake_pytest-extras.cmake) and it's not causing a build problem for us.

clalancette, 2019-09-24:
See https://github.com/ament/ament_cmake/pull/200; we are looking to add it to `ament_cmake_pytest`.

The basic reason it is needed is that when building Windows with `-DMAKE_BUILD_TYPE=Debug`, it's a whole separate stack of things.  That means it uses a separate python executable (`python_d.exe`), and a completely separate installation of pip packages.  Thus, in order for Python tests to be run in debug mode on Windows, we need that PYTHON_EXECUTABLE_DEBUG.

hidmic, 2019-09-24:
@clalancette statement is accurate. And FYI, `python_cmake_module` recently got split into its own repository. See https://github.com/ros2/python_cmake_module.

pbaughman, 2019-09-24:
Hum, Ok... I'll look into why we're having trouble with this package before the problem gets worse and we can no longer build things that rely on ament_cmake_pytest

I suspect it's:
```
find_package(PythonInterp 3.5 REQUIRED)
```

[from here](https://github.com/ros2/python_cmake_module/blob/master/cmake/Modules/FindPythonExtra.cmake#L45)

My understanding is we have a python interpreter at test time, but not at build time for some reason.  I'll continue to dig on our end.  Thanks guys!

pbaughman, 2019-09-24:
Ok - I think this is an internal-only issue where we're over-zealously dropping an AMENT_IGNORE on python_cmake_module because in our version it's still inside rosidl_python.  Thanks for your help understanding this package and the history of why it's here.

closed 2019-09-24 by pbaughman