ros2/launch #347: ExecuteProcess not returning on_exit actions in describe_conditional_sub_entities

pbaughman, 2019-10-23:
## Feature request
[Launch actions](https://github.com/ros2/launch/blob/f73bc56caa9c7b3c53b264475728392d0e1ce9cf/launch/launch/action.py#L91-L102) are supposed to return their sub-entities and conditional sub-entities to so they can be introspected before running.

The [ExecuteProcess action](https://github.com/ros2/launch/blob/master/launch/launch/actions/execute_process.py) does not provide an implementation of these methods, so anything run on_exit can not be introspected

#### Feature description

Implement `get_sub_entities` in ExecuteProcess so that we can discover on_exit actions.  One place where this is causing a problem right now is [here](https://github.com/ros2/ros2cli/blob/c61418e624cd7fb98ea8c521d2feb83113eb8232/ros2service/test/test_cli.py#L90)  launch_testing can't discover the ReadyToTest action (or even the `daemon-start' process) because it's nested down in on_exit

#### Implementation considerations

???

hidmic, 2019-11-12:
Looks like I inadvertently fixed this in https://github.com/ros2/launch/pull/353. Closing.

closed 2019-11-12 by hidmic

pbaughman, 2019-11-12:
@hidmic I had an in person conversation with @wjwwood where he suggested the sub-entities needed to include [the event handlers and events](https://github.com/ros2/launch/blob/2b278b0586f83c61ac0d7704e018a707bacf437d/launch/launch/actions/execute_process.py#L683-L709) that are plumbed into the launch context when  you call 'execute'

It's possible that you've un-jammed [my PR here](https://github.com/ros2/launch/pull/346) though (for which you're a wonderful human being and I thank you) but maybe William can chime in about whether or not we should keep this issue open and I can fill in the "Implementation considerations" section above a bit better

hidmic, 2019-11-12:
> @hidmic I had an in person conversation with @wjwwood where he suggested the sub-entities needed to include the event handlers and events that are plumbed into the launch context when you call 'execute'

I see, I wasn't aware. I do not have a strong opinion, though those event handlers seem an implementation detail to me but we haven't really drawn a line between what's introspectable and what not (or if such a line should even exist).

pbaughman, 2019-11-12:
Yeah, I don't feel strongly either.  I think William's idea was that the 'Register Event Handler' entities were unconditional sub-entities of an execute-process, and then the on-process-exit actions would be conditional sub-entities of THOSE entities.

I think you've fixed the problem I had in my head when I opened this issue.  One of the reasons it sat open was that the solution I was considering was much more complicated.

wjwwood, 2019-11-12:
>  I think William's idea was that the 'Register Event Handler' entities were unconditional sub-entities of an execute-process, and then the on-process-exit actions would be conditional sub-entities of THOSE entities.

That's right.