ros2/launch #364: I/O operation on closed file in __on_process_stderr

pbaughman, 2019-12-12:
## Bug report

**Required Info:**

- Operating System:
  - 18.04
- Installation type:
  - source
- Version or commit hash:
  - c644dfa8e1f1433e200b76eb52ba22f392838f48
- DDS implementation:
  - RTI Connect Micro, but I don't think this part matters
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

I can get this to happen with the rcl test "test_rmw_impl_id_check" if also running stress on the system

```
terminal 1> stress -m 6 -c 6
terminal 2> launch_test build/rcl/test/test/test_rmw_impl_id_check__rmw_apex_dds_RelWithDebInfo.py --verbose
. . .
. . .
[DEBUG] [launch]: Traceback (most recent call last):
  File "/home/pete.baughman/gc/apex_ws/build/launch/launch/launch_service.py", line 377, in run_async
    await process_one_event_task
  File "/usr/lib/python3.6/asyncio/coroutines.py", line 126, in send
    return self.gen.send(value)
  File "/home/pete.baughman/gc/apex_ws/build/launch/launch/launch_service.py", line 287, in _process_one_event
    await self.__process_event(next_event)
  File "/usr/lib/python3.6/asyncio/coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "/home/pete.baughman/gc/apex_ws/build/launch/launch/launch_service.py", line 296, in __process_event
    entities = event_handler.handle(event, self.__context)
  File "/home/pete.baughman/gc/apex_ws/build/launch/launch/event_handlers/on_process_io.py", line 75, in handle
    return self.__on_stderr(event)
  File "/home/pete.baughman/gc/apex_ws/build/launch/launch/actions/execute_process.py", line 414, in __on_process_stderr
    self.__stderr_buffer.write(event.text.decode(errors='replace'))
ValueError: I/O operation on closed file
```

This happens when the execute_process object is trying to print 
```
[rcl]: Expected RMW implementation identifier of 'garbage' but instead found 'rmw_apex_dds', exiting with 102
```

#### Expected behavior
Launch Service should not die when trying to print to a buffer during shutdown

#### Actual behavior
It seems like this stderr message is racing with the shutdown of the launch service.

#### Additional information
  - I don't see where the __stderr_buffer gets closed yet - continuing to investigate
  - I'm also having trouble reproducing this in the nightly docker because I can't get rcl to build in there.  Continuing to investigate that too

pbaughman, 2019-12-12:
Follow-up:  I can get the exact same failure pretty easily in the nightly ROS2 docker image:

```
docker pull osrf/ros2:nightly
docker run --rm -it -v ~/rcl:/rcl osrf/ros2:nightly

# Inside docker now
cd /rcl
colcon build --packages-up-to rcl
launch_test build/rcl/test/test/test_rmw_impl_id_check__rmw_fastrtps_cpp_.py --verbose
. . .
. . .
Starting test run test_rmw_impl_id_check__rmw_fastrtps_cpp_.test_rmw_impl_id_check__rmw_fastrtps_cpp_.launch_tests[garbage, None]
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
[INFO] [launch]: All log files can be found below /root/.ros/log/2019-12-12-17-48-38-773724-e872607aa153-13078
[INFO] [launch]: Default logging verbosity is set to DEBUG
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7fe132267828>'
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7fe132267828>' ✓
 '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x7fe132b68cf8>'
test_process_terminates_in_a_finite_amount_of_time[garbage, None] (test_rmw_impl_id_check__rmw_fastrtps_cpp_.TestRMWImplementationID
Check)
Test that executable under test terminates in a finite amount of time. ... [INFO] [/rcl/build/rcl/test/test_rmw_impl_id_check_exe-3]
: process started with pid [13085]
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.process.ProcessStarted'
[ERROR] [/rcl/build/rcl/test/test_rmw_impl_id_check_exe-3]: process has died [pid 13085, exit code 1, cmd '/rcl/build/rcl/test/test_
rmw_impl_id_check_exe'].
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.process.ProcessExited'
[INFO] [python3-4]: process started with pid [13086]
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.process.ProcessStarted'
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.process.ProcessStderr'
[DEBUG] [launch]: processing event: '<launch.events.process.process_started.ProcessStarted object at 0x7fe132270978>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_started.ProcessStarted object at 0x7fe132270978>' ✓ '<launch.eve
nt_handlers.on_process_start.OnProcessStart object at 0x7fe1322675f8>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_started.ProcessStarted object at 0x7fe132270978>' ✓ '<launch.eve
nt_handlers.on_process_start.OnProcessStart object at 0x7fe132267588>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x7fe1322709b0>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x7fe1322709b0>' ✓ '<launch.event
_handlers.on_process_exit.OnProcessExit object at 0x7fe132267c88>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x7fe1322709b0>' ✓ '<launch.event
_handlers.on_process_exit.OnProcessExit object at 0x7fe132267c18>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_exited.ProcessExited object at 0x7fe1322709b0>' ✓ '<launch.event
_handlers.on_process_exit.OnProcessExit object at 0x7fe132267668>'
ok

----------------------------------------------------------------------
Ran 1 test in 0.029s

OK
[DEBUG] [launch]: processing event: '<launch.events.process.process_started.ProcessStarted object at 0x7fe132267cc0>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_started.ProcessStarted object at 0x7fe132267cc0>' ✓ '<launch.eve
nt_handlers.on_process_start.OnProcessStart object at 0x7fe1322675f8>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_started.ProcessStarted object at 0x7fe132267cc0>' ✓ '<launch.eve
nt_handlers.on_process_start.OnProcessStart object at 0x7fe132267588>'
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.Shutdown'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7fe1322709e8>'
[DEBUG] [launch]: processing event: '<launch.events.process.process_stderr.ProcessStderr object at 0x7fe1322709e8>' ✓ '<launch.event
_handlers.on_process_io.OnProcessIO object at 0x7fe132267ba8>'
[DEBUG] [launch]: Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.6/site-packages/launch/launch_service.py", line 381, in run_async
    await process_one_event_task
  File "/usr/lib/python3.6/asyncio/coroutines.py", line 126, in send
    return self.gen.send(value)
  File "/opt/ros/foxy/lib/python3.6/site-packages/launch/launch_service.py", line 291, in _process_one_event
    await self.__process_event(next_event)
  File "/usr/lib/python3.6/asyncio/coroutines.py", line 110, in __next__
    return self.gen.send(None)
  File "/opt/ros/foxy/lib/python3.6/site-packages/launch/launch_service.py", line 300, in __process_event
    entities = event_handler.handle(event, self.__context)
  File "/opt/ros/foxy/lib/python3.6/site-packages/launch/event_handlers/on_process_io.py", line 75, in handle
    return self.__on_stderr(event)
  File "/opt/ros/foxy/lib/python3.6/site-packages/launch/actions/execute_process.py", line 448, in __on_process_stderr
    self.__stderr_buffer.write(event.text.decode(errors='replace'))
ValueError: I/O operation on closed file

[ERROR] [launch]: Caught exception in launch (see debug for traceback): I/O operation on closed file
usage: launch_test [-h] [--package-name PACKAGE_NAME] [-v] [-s]
                   [--junit-xml XMLPATH]
                   launch_test_file [launch_arguments [launch_arguments ...]]
launch_test: error: object NoneType can't be used in 'await' expression
```

This particular test starts a process that runs, possibly generates some stderr output, and then exits very very quickly.  That's probably why it's exercising this race but other things aren't.

pbaughman, 2019-12-13:
Ah, I see - the race is with closing the buffer implicitly [here](https://github.com/ros2/launch/blob/master/launch/launch/actions/execute_process.py#L471-L476)

@ivanpauno It looks like this code was introduced in PR #255.  Do you have an opinion about how to resolve this race?  I see [here](https://github.com/ros2/launch/pull/255#discussion_r290484218) that there was some question as to whether or not closing the buffers was even necessary.

In the meantime, I'm going to look at the call stack for __flush_buffers to see if there's a way to force it to happen after the __on_process_stderr function will no longer be called

ivanpauno, 2019-12-13:
Not closing the buffers sound reasonable to me.
The only problem, is that that late output might never be printed, but crashing is worst.

I think we've already hit this before ... though I don't remember where the previous discussion is.

pbaughman, 2019-12-13:
Maybe there is a way to sequence things so that process get shut down, __on_process_stderr/stdout callbacks fire, then __flush_buffers is called?

ivanpauno, 2019-12-13:
> Maybe there is a way to sequence things so that process get shut down, __on_process_stderr/stdout callbacks fire, then __flush_buffers is called?

I don't see an easy way of ensuring that the buffers are flushed after all other events are processed, due to the asynchronous nature of how events are processed.
If you have a way of doing that, sounds like the best solution to me.

If not, how io events are handled after shutdown could be changed, avoiding buffering and printing directly (which can end up with split messages on Windows, but it's not that bad).

ivanpauno, 2019-12-13:
Well, I finally see a way of doing what you proposed. A method for registering `on exit` callbacks in the context could be added.
Those callbacks will be called when the launch service event loop finished running, ensuring that they are called after all the events.

The counterpart is complexity, the difference between `on shutdown` event handlers and those callbacks is subtle.

pbaughman, 2019-12-13:
@ivanpauno I will take a crack at

> How io events are handled after shutdown could be changed, avoiding buffering and printing directly (which can end up with split messages on Windows, but it's not that bad).

I think that's a good compromise between not losing log messages and complexity.  That should fix the crash and I don't the severity of the remaining problem justifies doing major surgery  on the launch context.

PR #365 (2019-12-16): Handle case where output buffer is closed during shutdown
  - Prevent crash during launch shutdown when a process IO event happens
    after the buffers have been closed

  - Use unbuffered output in that case so IO still has a chance of being seen

Fixes #364
closed 2019-12-16 by hidmic