ros2/launch #366: Add a Command substitution

ivanpauno, 2020-01-03:
## Feature request

#### Feature description

In ROS1, it was possible to load a parameter value with the output of a command:

```xml
<param name="robot_description" command="$(find xacro)/xacro --inorder
                       '$(find my_package)/path/to/xacro/file'"/>
```

In order to support similar functionality in ROS2 launch, a `Command` substitution can be added.
Adding a substitution instead of adding an specific way of loading a parameter value with a command output will allow other use cases.

Example usage:

```python
Node(
  node_executable = 'my_node',
  package='my_package',
  parameters=[{'robot_description': Command(
    [
      PathJoinSubstitution([FindPackagePrefix('xacro'), 'xacro']),
      ' --inorder ',
      PathJoinSubstitution([FindPackagePrefix('my_package'), 'path', 'to', 'xacro', 'file']),
    ]
  }]
)
```

#### Implementation considerations

Implementation skeleton:

```python3

class CommandSubstitution(Substitution):

  def __init__(self, command: SomeSubstitutionsType):
    ...

  def perform(self, context: LaunchContext) -> Text:
    ...
```

ivanpauno, 2020-01-03:
@wjwwood @hidmic can you review the proposal?

hidmic, 2020-01-03:
LGTM, though we should do something about ergonomics when this gets mapped to XML/YAML - - it's crazy verbose :sweat_smile:.

wjwwood, 2020-01-03:
lgtm too.

ivanpauno, 2020-01-09:
> LGTM, though we should do something about ergonomics when this gets mapped to XML/YAML - - it's crazy verbose .

About this: I think that most of the verbosity comes from the need of the `PathJoinSubstitution` stuff.
Forgetting about that, it would look in xml like:

```xml
<param name="robot_description" value="$(command '$(find_pkg_prefix xacro)/xacro --inorder
                       $(find my_package)/path/to/xacro/file'"/>
```

(note that the command is between quotes, because I think it's the most natural way to match what we do in Python)

That's not super verbose.
We could also have directly an `output` attribute for params like in ROS 1, to make it less verbose.
Another options is having a specific substitution for `xacro` (which is pretty common).

Paths are a problem here (if we want the launch file to work in all the platforms).
It's hard to accept posix paths here without adding a substitution.
The first part of the command (`$(find_pkg_prefix xacro)/xacro`) is known to be a path, so we can avoid a substitution there, and accept a posix path.
The problem is if there are other paths in the arguments of the command, a substitution is required there.

Instead of the `PathJoinSubstitution` (which wouldn't look nice in xml), we should probably have a `Path` one, that indicates that the passed string is a posix path.

Combining that, it will look in `xml` like:
```xml
<param name="robot_description" value="$(command '$(find_pkg_prefix xacro)/xacro --inorder
                       $(path $(find my_package)/path/to/xacro/file)'"/>
```

(note: `path` can also be used for the command itself, without doing any harm)
(note2: people that don't care about windows can forget about the `path` substitution)

I think the last example is not bad. Doing it less verbose would be really hard.

closed 2020-01-15 by ivanpauno

akela1101, 2022-08-16:
@ivanpauno 
Please, help me with documentation of this feature.
Should it be added to these places?

https://docs.ros.org/en/foxy/How-To-Guides/Launch-files-migration-guide.html
&
https://design.ros2.org/articles/roslaunch_xml.html

ivanpauno, 2022-08-16:
> https://docs.ros.org/en/foxy/How-To-Guides/Launch-files-migration-guide.html
> &
> https://design.ros2.org/articles/roslaunch_xml.html

Yeah, adding this to both articles seems fine.
Feel free to add me as a reviewer if you open a PR.